istoreos/istoreos #389: 请求加入Turbo ACC插件

eigfish, 2022-09-23:
This repository is a mirror of our main repo at https://git.openwrt.org/openwrt/openwrt.git
It is for reference only and is not active for checkins or for reporting issues.

All issues should be reported at: https://bugs.openwrt.org

Please do not open any NEW issue here - we will be closing the Github issues
capability in the near future.

We will continue to accept Pull Requests here as described in the repo description.

Close this now and click here: https://bugs.openwrt.org

Thankyou for your co-operation.

jannson, 2022-09-24:
测试不稳定，所以我们还一直没敢加。会增加很多客服，好些时候配置就莫名其妙断网。。。
自己安装也许是可以的

xzkyo, 2023-07-13:
为什么不加入？这个能翻倍提高小包转发

berryyan, 2023-11-12:
自己如何安装

Meilinhost, 2024-01-07:
有进展吗? 防火墙里的“基于软件的 路由/NAT 分载” 是不是 Shortcut-FE