istoreos/istoreos #393: 啥时候可以支持Nanopi R2C，期待这个固件好久了

hongmaju, 2022-09-27:
This repository is a mirror of our main repo at https://git.openwrt.org/openwrt/openwrt.git
It is for reference only and is not active for checkins or for reporting issues.

All issues should be reported at: https://bugs.openwrt.org

Please do not open any NEW issue here - we will be closing the Github issues
capability in the near future.

We will continue to accept Pull Requests here as described in the repo description.

Close this now and click here: https://bugs.openwrt.org

Thankyou for your co-operation.

hongmaju, 2022-09-28:
[image]

hongmaju, 2022-09-28:
wan口，YT8521S的网卡驱动

GolemZ2K, 2022-10-29:
卖了，换R2S，马上就能用上！

jjm2473, 2022-12-16:
不兼容r2s的话就算了

closed 2022-12-16 by jjm2473

jjm2473, 2022-12-19:
最新r2s固件已经集成yt8521驱动，r2c直接刷r2s固件就行了