istoreos/istoreos #394: 大宝哥 在挂载网络共享里 把nas文件夹挂载到op 能成功 并被jellyfin读取 可以过一阵子路径会自动消失不见 再次开启共享又出现 就是待不久 机器正常运行 无任何操作 消失了 jellyfin也读不到媒体文件夹了

molin830, 2022-09-27:
[This](url) repository is a mirror of our main repo at https://git.openwrt.org/openwrt/openwrt.git
It is for reference only and is not active for checkins [or]([image]) for reporting issues.

All issues should be reported at: https://bugs.openwrt.org

Please do not open any NEW issue here - we will be closing the Github issues
capability in the near future.

We will continue to accept Pull Requests here as described in the repo description.

Close this now and click here: https://bugs.openwrt.org

Thankyou for your co-operation.

molin830, 2022-09-27:
[image]
[image]
[image]

My-By, 2022-11-29:
不知道你解决了没有，我今天刚解决，把挂载点设置成全英文，不带中文的，就算掉了也会自动挂载进去，相对来说可能会好一点。

My-By, 2022-12-01:
[image]
[image]
亲测只要是中文挂载点很容易多设备接入或者短暂延迟断网就会掉挂载，一旦全英文配置，重启，掉挂载也会自动挂载上去，很稳定。

jjm2473, 2023-01-10:
是不是网络不稳定，还有不要把路径设置到nvme什么里面，直接/mnt/nas不行吗？

My-By, 2023-02-06:
在网络共享里填5次你要挂载的网络地址 五条填一模一样的，我的E3845填一次稳如老狗，J4125填一次重启或者断网就掉，无法自动挂载，填5次就很稳，具体原因不明。

HFfay, 2023-02-07:
有挂载  WebDAV 成功的么？应用之后，挂载目录生成了，但是里面是空的。
[image]
[image]

jjm2473, 2023-02-07:
@HFfay 试试完整的url，http开头的

HFfay, 2023-02-08:
> 

谢谢，还是没有用。

主要是没教程，网上也找不到相关配置的例子。

HFfay, 2023-02-09:
我这边解决了，需要在url后面再加一层/dav/才行

jjm2473, 2023-02-10:
> 我这边解决了，需要在alist的url后面再加一层/dav/才行 完整url是这样的：[http://ip:端口/dav/](http://ip:%E7%AB%AF%E5%8F%A3/dav/)

我靠，alist后台页面不是告诉你webdav的路径了吗，居然会不知道要家dav？？

closed 2023-02-10 by jjm2473

Joker-lm, 2023-04-27:
[image]
用户那里只能选guest,自定义没用啊，挂不成功

auroracamus, 2023-05-04:
> 在网络共享里填5次你要挂载的网络地址 五条填一模一样的，我的E3845填一次稳如老狗，J4125填一次重启或者断网就掉，无法自动挂载，填5次就很稳，具体原因不明。

亲测这个方法好用，大力出奇迹~

Joker-lm, 2023-05-06:
> 

用户那里不能选，怎么解决的，兄弟

Joker-lm, 2023-05-06:
> > 在网络共享里填5次你要挂载的网络地址 五条填一模一样的，我的E3845填一次稳如老狗，J4125填一次重启或者断网就掉，无法自动挂载，填5次就很稳，具体原因不明。
> 
> 亲测这个方法好用，大力出奇迹~

用户那里不能选，不能用admin账户，怎么解决的，兄弟

Joker-lm, 2023-05-09:
> @Joker-lm 用户那下拉不是可以自己填吗，填完回车

囧😳，我没回车

411EN, 2024-03-18:
> 我这边解决了，需要在alist的url后面再加一层/dav/才行 完整url是这样的：[http://ip:端口/dav/](http://ip:%E7%AB%AF%E5%8F%A3/dav/)

我也是填写完整路径，挂载的文件夹里还是空的 请问怎么解决

houuu23, 2024-04-28:
必须要用户名跟密码

wuchjun, 2025-04-26:
[image]
填写了账号密码，访问目录提示“登录状态丢失”

> 必须要用户名跟密码