istoreos/istoreos #397: 你好，最新版istoreos-21.02.3-2022092414-x86-64不能安装软件包

Rickyzhu1, 2022-09-28:
安装任何软件包，都提示以下错误，请问怎么解决？

正在执行软件包管理器
Installing ruby (3.0.4-1) to root...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/x86_64/packages/ruby_3.0.4-1_x86_64.ipk

错误
Error loading shared library libwolfssl.so.5.2.0.99a5b54a: No such file or directory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
Collected errors:
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/x86_64/packages/ruby_3.0.4-1_x86_64.ipk, curl returned 127.
 * opkg_install_pkg: Failed to download ruby. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package ruby.

opkg install 命令失败，代码 255。

binsir1, 2022-10-08:
和你一样

kid-oh, 2022-11-06:
> 到 QQ 群里用 reset_rom_xxx.run 恢复下系统的 curl 这个软件包，它被之前你安装的包替换了。。。

老哥有下载连接分享一下吗

hzgotb, 2022-11-08:
wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/curl_7.83.1-1_x86_64.ipk
wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/libcurl4_7.83.1-1_x86_64.ipk
wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/base/libwolfssl5.2.0.99a5b54a_5.2.0-stable-2_x86_64.ipk
opkg install --force-reinstall libwolfssl5.2.0.99a5b54a_5.2.0-stable-2_x86_64.ipk
opkg install --force-reinstall libcurl4_7.83.1-1_x86_64.ipk
opkg install --force-reinstall curl_7.83.1-1_x86_64.ipk

[19547](https://github.com/openwrt/packages/issues/19547#issuecomment-1279921944)

nevertoday, 2022-12-07:
r4s 怎么恢复，请问