istoreos/istoreos #400: istore安装应用报错

mikutechnology, 2022-10-03:
在使用istore不管安装什么应用，统一报错：
Error loading shared library libwolfssl.so.5.2.0.99a5b54a: No such file or directory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
软件源用的是官方源，重装过系统也一样

expliyh, 2022-10-03:
我也是，从手动安装也有这种错误

expliyh, 2022-10-03:
是不是用过了shellclash？

expliyh, 2022-10-03:
貌似是用过了 opkg update 以后就会出现

ldpc520, 2022-10-04:
无论怎么操作都无法安装其他插件啊！这样就没意思了

mikutechnology, 2022-10-04:
已经解决了，虽然有些麻烦，我来说一下怎么解决的吧
1.打开软件包-配置OPKG-先把option use_curl 1删除掉。点保存
2.重新运行一次opkg update
3.运行opkg install libcurl4 curl，等待安装完成
4.再次打开配置OPKG，把option use_curl 1加回去点保存，然后再运行opkg update，回到iStore，随便安装一个插件试试，不出所料就能恢复安装了

expliyh, 2022-10-09:
> [https://op.supes.top/packages/x86_64/我在这下了一个科学上网的插件，按照你的说明去操作最后还是安装失败了](https://op.supes.top/packages/x86_64/%E6%88%91%E5%9C%A8%E8%BF%99%E4%B8%8B%E4%BA%86%E4%B8%80%E4%B8%AA%E7%A7%91%E5%AD%A6%E4%B8%8A%E7%BD%91%E7%9A%84%E6%8F%92%E4%BB%B6%EF%BC%8C%E6%8C%89%E7%85%A7%E4%BD%A0%E7%9A%84%E8%AF%B4%E6%98%8E%E5%8E%BB%E6%93%8D%E4%BD%9C%E6%9C%80%E5%90%8E%E8%BF%98%E6%98%AF%E5%AE%89%E8%A3%85%E5%A4%B1%E8%B4%A5%E4%BA%86) ??健少?? ***@***.***  
> […](#)

不用把那一行加回去，直接不用curl

kid-oh, 2022-11-05:
这reset.run不能在github上也放一份？要的时候还得进qq群

kiwi5656, 2023-09-16:
> 请到群里，用 reset_rom_xxx.run 安装并重启，然后恢复系统软件包。你安装了其他插件，把系统软件包给替换了，就出问题了。

群怎么进？