istoreos/istoreos #406: 系统从0560稳定版升级，NFS文件系统太慢。

lingyunzhiss, 2022-10-15:
系统从0560稳定版升到最高0924版，硬件和布线没做改动。原想着提升便利性和看有没有性能提升的。使用系统提供的NFS发现同样的影片在KODI变卡顿。立即着手调整，因为0560版是纯手动挂载的，4K影片58G的拯救大兵毫无压力，动手想恢复手动挂载。发现NFS实现方法不一样。原来的我OPKG下卸载了，现在是4个包：
root@Openwrt:~# opkg install nfs-kernel-server
Installing nfs-kernel-server (2.5.2-1) to root...
Downloading https://downloads.openwrt.org/sn ... _2.5.2-1_x86_64.ipk
Installing libkeyutils1 (1.6.3-1) to root...
Downloading https://downloads.openwrt.org/sn ... _1.6.3-1_x86_64.ipk
Installing libdevmapper-selinux (2.03.11-1) to root...
Downloading https://downloads.openwrt.org/sn ... .03.11-1_x86_64.ipk
Installing nfs-utils-libs (2.5.2-1) to root...
Downloading https://downloads.openwrt.org/sn ... _2.5.2-1_x86_64.ipk
Configuring libkeyutils1.
Configuring libdevmapper-selinux.
Configuring nfs-utils-libs.
Configuring nfs-kernel-server.
这一套下来NFS飞快。现在手动安装出来NFS卡得不行1080P的片子都要停下来缓冲。
1）我还能恢复LUCI-app-nfs，luci-i8n-nfs-zh_cn的安装吗？
2）在iStoreOS 21.02.3 2022092414下有没有更好的NFS方案？

lingyunzhiss, 2022-10-15:
问题已经解决了，跟nfs无关，是挂载点的问题。如图可以达到最高速80M/S
[image]

closed 2022-10-15 by lingyunzhiss

jjm2473, 2024-07-02:
截图是samba的配置，跟NFS没任何关系