istoreos/istoreos #409: 在更换软件源后一直报错，OPKG更新的时候 大量报错

w2816771, 2022-10-20:
在更换软件源后一直报错，切换回原有的软件源后，依旧如此，尝试执行 opkg list-upgradable | awk -F ' - ' '{print $1}' | xargs opkg upgrade 时候，依旧有大量报错。 
我尝试了备份重置中的恢复出厂设置，结果一样

请问大佬如何解决

``` BASH
Upgrading odhcpd-ipv6only on root from 2021-07-18-bc9d317f-1 to 2021-07-18-bc9d3                                                                                                                        17f-3...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packag                                                                                                                        es/x86_64/base/odhcpd-ipv6only_2021-07-18-bc9d317f-3_x86_64.ipk
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or direc                                                                                                                        tory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
Upgrading libnetwork on root from 2021-05-25-64b7a457-2 to 2022-06-02-f6ccccb1-1                                                                                                                        ...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packag                                                                                                                        es/x86_64/packages/libnetwork_2022-06-02-f6ccccb1-1_x86_64.ipk
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or direc                                                                                                                        tory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
Upgrading libiwinfo-data on root from 2021-04-30-c45f0b58-2.1 to 2022-04-26-dc68                                                                                                                        47eb-1...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packag                                                                                                                        es/x86_64/base/libiwinfo-data_2022-04-26-dc6847eb-1_x86_64.ipk
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or direc                                                                                                                        tory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
Upgrading luci-base on root from git-22.258.47264-284140f to git-22.288.45155-af                                                                                                                        d0012...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packag                                                                                                                        es/x86_64/luci/luci-base_git-22.288.45155-afd0012_x86_64.ipk
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or direc                                                                                                                        tory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
Upgrading libopenssl1.1 on root from 1.1.1n-1 to 1.1.1q-1...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packag                                                                                                                        es/x86_64/base/libopenssl1.1_1.1.1q-1_x86_64.ipk
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or direc                                                                                                                        tory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
Upgrading luci-i18n-ddns-zh-cn on root from git-22.258.47264-284140f to git-22.2                                                                                                                        77.72334-3923199...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packag                                                                                                                        es/x86_64/luci/luci-i18n-ddns-zh-cn_git-22.277.72334-3923199_all.ipk
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or direc                                                                                                                        tory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
Upgrading libopenssl-conf on root from 1.1.1n-1 to 1.1.1q-1...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packag                                                                                                                        es/x86_64/base/libopenssl-conf_1.1.1q-1_x86_64.ipk
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or direc                                                                                                                        tory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
Upgrading containerd on root from 1.4.12-1 to 1.6.6-1...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packag                                                                                                                        es/x86_64/packages/containerd_1.6.6-1_x86_64.ipk
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or direc                                                                                                                        tory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
Upgrading libiwinfo20210430 on root from 2021-04-30-c45f0b58-2.1 to 2022-04-26-d                                                                                                                        c6847eb-1...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packag                                                                                                                        es/x86_64/base/libiwinfo20210430_2022-04-26-dc6847eb-1_x86_64.ipk
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or direc                                                                                                                        tory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
Upgrading hostapd on root from 2020-06-08-5a8b3662-40 to 2020-06-08-5a8b3662-41.                                                                                                                        ..
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packag                                                                                                                        es/x86_64/base/hostapd_2020-06-08-5a8b3662-41_x86_64.ipk
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or direc                                                                                                                        tory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
Upgrading runc on root from 1.0.2-1 to 1.1.2-1...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packag                                                                                                                        es/x86_64/packages/runc_1.1.2-1_x86_64.ipk
[... cut ...]
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwr                                                                                                                        t/releases/21.02.3/packages/x86_64/base/dnsmasq-full_2.85-8_x86_64.ipk, curl ret                                                                                                                        urned 127.
 * opkg_install_pkg: Failed to download dnsmasq-full. Perhaps you need to run 'o                                                                                                                        pkg update'?
xargs: opkg: exited with status 255; aborting
root@iStoreOS:~#
```

jjm2473, 2022-10-23:
不要乱更新，可能会更新到不兼容的版本，尤其是不兼容的库文件，可能直接让很多程序无法运行。安装reset_rom_pkgs把一些软件恢复到固件版本就好了，或者手动安装curl以及它的依赖来修复，不然现在下载ipk都会失败

w2816771, 2022-10-23:
感谢，重装了系统解决了

closed 2022-10-23 by w2816771

nevertoday, 2022-12-07:
我不想重装系统，呜呜 我配置了太多docker 了， 如何....

jjm2473, 2022-12-08:
> 我不想重装系统，呜呜 我配置了太多docker 了， 如何....

更新到11月28号以后的固件，curl 可以自动回避这种错误

nevertoday, 2022-12-08:
用了， 问题依旧， 重新安装系统了， 有没有办法很好的备份自己的配置，我好多docker 全部在系统备份压缩包中，恢复后，全部蒸发

jjm2473, 2022-12-19:
升级固件以后还是同样的错误？