istoreos/istoreos #419: R2S的2022102109版本frpc，frps，transmission插件有问题

GolemZ2K, 2022-10-28:
R2S的2022102109版本frpc，frps安装出错，transmission插件启用之后，打不开9091端口。

GolemZ2K, 2022-10-28:
[image]

jjm2473, 2023-02-23:
transmission看看启动没，启动了的话界面上有提示。
RPC错误什么的，使用r68s测试没发现，可能已经修复

appdev, 2023-02-25:
并没有修复，R2S昨天刚下载的最新版本，配置了路由模式后什么都没有动，直接安装 frpc frps 都可以稳定复现这个问题。

jjm2473, 2023-02-25:
配置旁路由？

appdev, 2023-02-25:
跟是不是旁路由没关系，主路由旁路油都是这样

jjm2473, 2023-02-25:
配置路由模式是在首页那边配置了拨号吗，然后安装就有问题？

appdev, 2023-02-25:
是的。只安装了frpc

jjm2473, 2023-02-25:
如果作为二级路由，不配置拨号的情况下，会有这个问题吗

appdev, 2023-02-25:
刚才试了一下，把它当作旁路有使用，还是有这个问题。这个问题应该挺久了吧。去年12月的时候就有这个问题。
[image]

appdev, 2023-02-25:
我猜应该是`luci-i18n-frpc-zh-cn` 这个软件包有问题。从 iStore 安装的版本会默认装上 `luci-app-frpc` `luci-i18n-frpc-zh-cn` 我卸载了iStore 安装的 frp 通过 opkg  安装  `luci-app-frpc` 就正常了。也是中文的

jjm2473, 2023-02-25:
@appdev 你在istore里面安装一下frpc，然后复制安装日志发出来看看

appdev, 2023-02-25:
```shell
Installing luci-app-frpc (git-21.175.36114-ad08c8e) to root...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/luci/luci-app-frpc_git-21.175.36114-ad08c8e_all.ipk
Installing frpc (0.34.3-1) to root...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/packages/frpc_0.34.3-1_aarch64_generic.ipk
Configuring frpc.
Configuring luci-app-frpc.
```

appdev, 2023-02-25:
哦 刚才发错了。刚才是 opkg 的。这是iStore的：
Installing app-meta-frpc (1.0.0-2) to root...
Downloading https://istore.linkease.com/repo/all/meta/app-meta-frpc_1.0.0-2_all.ipk
Installing frpc (0.34.3-1) to root...
Downloading https://istore.linkease.com/repo/aarch64_cortex-a53/nas/frpc_0.34.3-1_aarch64_cortex-a53.ipk
Installing luci-app-frpc (git-21.175.36114-ad08c8e) to root...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/luci/luci-app-frpc_git-21.175.36114-ad08c8e_all.ipk
Installing luci-i18n-frpc-zh-cn (git-22.362.26829-d548d85) to root...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/luci/luci-i18n-frpc-zh-cn_git-22.362.26829-d548d85_all.ipk
Configuring frpc.
Configuring luci-app-frpc.
Configuring luci-i18n-frpc-zh-cn.
Configuring app-meta-frpc.

jjm2473, 2023-02-25:
我用x86安装了多个版本的luci-app-frpc，都没出问题。理论上，luci程序都是一样的，因为架构无关，如果r2s有问题，其他设备也应该有问题。你试试chrome浏览器有没有报错

jjm2473, 2023-02-25:
哦，我看他们安装的frpc可执行程序不一样。试试istore安装以后，执行 `opkg install --force-reinstall frpc`，之后再看有没有错误

jjm2473, 2023-02-25:
还有执行 `ubus call service list '{"name":"frpc"}'` 看看有没有报错

appdev, 2023-02-25:
> 哦，我看他们安装的frpc可执行程序不一样。试试istore安装以后，执行 `opkg install --force-reinstall frpc`，之后再看有没有错误

试了两次这么装还是报错。

[image]

appdev, 2023-02-25:
> 我猜应该是`luci-i18n-frpc-zh-cn` 这个软件包有问题。从 iStore 安装的版本会默认装上 `luci-app-frpc` `luci-i18n-frpc-zh-cn` 我卸载了iStore 安装的 frp 通过 opkg 安装 `luci-app-frpc` 就正常了。也是中文的

这么装是正常的：
- 通过 opkg 安装 luci-app-frpc 这时候是英文版的
- 通过 iStore 安装 frpc 
这么装就是正常的，而且也变成中文了。

```shell
Installing app-meta-frpc (1.0.0-2) to root...
Downloading https://istore.linkease.com/repo/all/meta/app-meta-frpc_1.0.0-2_all.ipk
Installing luci-i18n-frpc-zh-cn (git-22.362.26829-d548d85) to root...
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/luci/luci-i18n-frpc-zh-cn_git-22.362.26829-d548d85_all.ipk
Configuring luci-i18n-frpc-zh-cn.
Configuring app-meta-frpc.
```

appdev, 2023-02-25:
对比了一下，应该就是执行文件不一样

jjm2473, 2023-02-25:
执行 `frpc -v` 看看有没有异常，如果正常显示版本号，那就再执行 `/etc/init.d/frpc restart ; ubus call service list '{"name":"frpc"}'` 看看

appdev, 2023-02-26:
现在正常了。

appdev, 2023-02-26:
原来只是UI看着正常了。在保存的配置的时候偶尔还是会出现 
[image]

jjm2473, 2023-02-28:
之后我找个r2s来测试吧

closed 2023-04-06 by jjm2473