istoreos/istoreos #427: istoreOS install error root not found

cgy0214, 2022-11-01:
u盘放在机器上执行安装系统时，出现错误求助。
启动正常跑码
执行到30.617376 NFSD: starting  10-second grace period (net f0000097) 最后一直停留再此。
1.回车出现root目录
2.执行 quickstart 出现安装菜单
3.执行 install X86
报错异常：error:root not found 
查看网卡信息选项为空。

反复试过的版本：
istoreos-21.02.1-2022042317-x86-64-squashfs-combined.img.gz
istoreos-21.02.1-2022041121-x86_64-squashfs-combined.img.gz
istoreos-21.02.1-2022041121-x86-64-generic-squashfs-combined-efi.img.gz
istoreos-21.02.1-2022042317-x86-64-generic-squashfs-combined-efi.img.gz

cgy0214, 2022-11-01:
是识别不到安装的U盘，还是机器里的固态硬盘呀？。机器里有两块我已经卸掉一块了，把u盘也设置第一启动项了，还是不行。有什么解决办法吗

closed 2022-11-07 by cgy0214

cgy0214, 2022-11-17:
直接盘刷就好了,不用这个包的。

cu97ming, 2022-12-23:
> 直接盘刷就好了,不用这个包的。

请问下 直接盘刷是指直接把img镜像写入到磁盘中吗？

likooo12580, 2023-05-05:
有人解决这个问题了么？ 求点步骤（来个英文教程网址也行）

cgy0214, 2023-07-28:
使用physdiskwrite用PE进把磁盘重新删分区然后用cmd执行 phydiskwrite.exe -u xxxx-efi.img 进入窗口后一定要执行 mount -o remount rw /不然没有权限保存不了配置， 新盘不会有这个问题也可以直接换个硬盘

cgy0214, 2023-07-28:
> > 直接盘刷就好了,不用这个包的。
> 
> 请问下 直接盘刷是指直接把img镜像写入到磁盘中吗？
使用physdiskwrite用PE进把磁盘重新删分区然后用cmd执行 phydiskwrite.exe -u xxxx-efi.img 进入窗口后一定要执行 mount -o remount rw /不然没有权限保存不了配置， 新盘不会有这个问题也可以直接换个硬盘

cgy0214, 2023-07-28:
> 有人解决这个问题了么？ 求点步骤（来个英文教程网址也行）

见上