istoreos/istoreos #431: 使用ventoy打开istoreos的ISO和img都会卡在new full-speed USB device number 4 using ci_hdrc

keng3333, 2022-11-02:
This repository is a mirror of our main repo at https://git.openwrt.org/openwrt/openwrt.git
It is for reference only and is not active for checkins or for reporting issues.

All issues should be reported at: https://bugs.openwrt.org

Please do not open any NEW issue here - we will be closing the Github issues
capability in the near future.

We will continue to accept Pull Requests here as described in the repo description.

Close this now and click here: https://bugs.openwrt.org

Thankyou for your co-operation.

jjm2473, 2023-01-04:
看看是不是跟这个有关 https://www.ventoy.net/cn/doc_openwrt.html

jjm2473, 2023-05-12:
已经修复了

jjm2473, 2023-05-12:
之后的固件支持ventoy启动，而且不需要按官方文档的特殊处理，解压后就能启动，还支持升级和安装到其他设备上

closed 2023-05-12 by jjm2473