istoreos/istoreos #436: iStoreOS 系统没有高级配置

zhouyunjian, 2022-11-05:
This repository is a mirror of our main repo at https://git.openwrt.org/openwrt/openwrt.git
It is for reference only and is not active for checkins or for reporting issues.

All issues should be reported at: https://bugs.openwrt.org

Please do not open any NEW issue here - we will be closing the Github issues
capability in the near future.

We will continue to accept Pull Requests here as described in the repo description.

Close this now and click here: https://bugs.openwrt.org

Thankyou for your co-operation.

原LEDE_X64_fw867/ 固件 系统 页有高级配置可以直接在web页面修改下边的配置文件,iStoreOS取消了这个功能希望能加上
--
/etc/dnsmasq.conf
/etc/config/network
/etc/config/DHCP
/etc/config/firewall
/etc/rc.local 

[LEDE_X64_fw867/](https://fw.koolcenter.com/LEDE_X64_fw867/)

zhouyunjian, 2022-11-14:
> 

出于对用户友好的方面，当然可以ssh到系统里修改，只是之前LEDE把这些常用的放在配置里边还是比较方便的，确实是个小优化，看你们觉得有没有必要了。

nevertoday, 2022-12-09:
请问要怎么设置成为，可以外网访问？？？