istoreos/istoreos #439: 什么时候出R2C的固件呀，现在我R2C只能当单臂路由来用

HAIHAODIE, 2022-11-07:
This repository is a mirror of our main repo at https://git.openwrt.org/openwrt/openwrt.git
It is for reference only and is not active for checkins or for reporting issues.

All issues should be reported at: https://bugs.openwrt.org

Please do not open any NEW issue here - we will be closing the Github issues
capability in the near future.

We will continue to accept Pull Requests here as described in the repo description.

Close this now and click here: https://bugs.openwrt.org

Thankyou for your co-operation.

jjm2473, 2022-12-19:
最新的r2s固件集成了yt8521驱动，看看能不能用了

HAIHAODIE, 2022-12-19:
> 跟r2s固件兼容吗，不兼容就放弃了

r2s是没问题的

jjm2473, 2022-12-19:
就是r2c刷r2s的固件啊，之前哪有单独出r2c的固件

jjm2473, 2022-12-19:
新的r2s固件集成了yt8521网卡驱动，应该支持r2c的lan口了

HAIHAODIE, 2022-12-19:
晚上我试试看，现在我刷骷髅的固件在用，但是🪜极不稳定，之前用istoreos的时候除了单网口其他基本没问题

HAIHAODIE, 2022-12-19:
> OK，我试过了，他这个固件并不是说不能用，lan口一直可以用但是wan口死活起不来，但是用其他固件又没问题，我还是先用其他固件吧，看后面会不会整个r2c的固件

jjm2473, 2023-01-30:
尝试让内核自动判断网卡芯片，试试这个固件 https://github.com/istoreos/istoreos/issues/566

jjm2473, 2023-02-05:
已发布，刷r2s固件即可

closed 2023-02-05 by jjm2473