istoreos/istoreos #458: docker 根目录重新挂载之后无法启动docker了

wangwendong1024, 2022-11-15:
[image]
docker 根目录重新挂载之后无法启动docker

My-By, 2022-12-01:
试试systools工具集修复系统软件。

wangwendong1024, 2022-12-01:
尝试了 不好使

发自我的 iPhone

My-By, 2022-12-01:
我也出现过一次，后来只能重装了。

hangaj, 2022-12-03:
出现同样的问题，挂到opt目录可以启动， 挂到外置硬盘就不行了，请大佬解答下哈

My-By, 2022-12-03:
我是刚装系统的时候第一次挂载也可挂载到sata接的盘上，其他的没试过，用一段时间换挂载点好像就会出问题，docker里好像有镜像和容器想要转挂载就出问题，

hangaj, 2022-12-03:
之前还出现过docker容量，重启后丢失的情况。。。哎，

My-By, 2022-12-03:
没有docker修复的功能确实不好整，小白啥都不会，

hangaj, 2022-12-03:
主要是不稳定。。各种问题。。。

My-By, 2022-12-03:
得看你是什么配置的机器了，我X86的E3845,4G内存，128G固态和一台J4125，16G内存，16G固态外接sata1T固态，用起来其实还好，目前没出现过不稳定的情况，就是偶尔有点小问题，最惨重装一下都能解决。

My-By, 2022-12-03:
docker是最近几个月才开始学着用的，刚开始就试试网心云，最近用了用nastools这些的，用起来还可以。

hangaj, 2022-12-03:
这是路由啊，哪里可能天天在那重装，被家里人骂死

My-By, 2022-12-03:
那就用旁路由模式呗，除非你有多拨需求，我软路由都用旁路由模式，除了多拨其他没什么区别，出现故障有很好维修，除非你是吧软路由当成wifi路由使。

hangaj, 2022-12-03:
旁路由，翻墙会有啥影响没

My-By, 2022-12-03:
如果网关互指的话是没什么区别的，比如你的主路由网关是192.168.1.1，旁路由是192.168.2.1，旁路的填主路由的，主路由的填旁路的，这样翻墙的话就没什么区别，如果软路由出问题了，主路由的网关删掉保存重启一下其他设备正常上网。如果是你自己用别人用不到的话，电脑，手机等设备自定义网关也可以。你要想问的话可以加我VX   Ch1neseMed1c1ne，我可以教你。

hangaj, 2022-12-03:
应用过滤在旁路由模式能不能正常工作

My-By, 2022-12-03:
应用过滤我没用过，那你得查查看了，

My-By, 2022-12-03:
我查了一下看到是mac地址过滤的，那应该是可以用的，

hangaj, 2022-12-03:
好，感谢回答！

My-By, 2022-12-03:
不过要应该过滤得让流量跑一遍openwrt，所以一般是得互指网关的，也就是说你在维修重装openwrt的时候，虽然可以上网，但是因为没有插件加成，流量是只经过主路的，所以那段时间是普通流量，不过你都在维护了，也不差那一点半点时间，

hangaj, 2022-12-03:
在软件包更新下 docker就能正常更换目录了。。。