istoreos/istoreos #459: 希望可以增加对rtl8111系列网卡的支持

ggg-g, 2022-11-15:
目前默认使用的是r8169这个驱动，瑞昱官方有r8168的驱动，希望可以增加对rtl8111系列网卡的支持

ggg-g, 2022-12-03:
发了好久了，希望被看到

ggg-g, 2022-12-13:
@jjm2473 @jannson

jjm2473, 2022-12-15:
OK，我试试能不能编译

jjm2473, 2022-12-15:
@ggg-g 哪些机型可以测试这个驱动？

ggg-g, 2022-12-15:
> @ggg-g 哪些机型可以测试这个驱动？

我的是nuc改的双网卡。
我在恩山那边也看到不少X86用这款卡的。
lean大已经将openwrt自带的r8169取消默认，而是换成了r8168

jjm2473, 2022-12-15:
我没有机器测试，等下我把ipk发上来给你测试

jjm2473, 2022-12-15:
[kmod-r8168_5.4.188+8.051.02-1_x86_64.ipk.zip](https://github.com/istoreos/istoreos/files/10235512/kmod-r8168_5.4.188%2B8.051.02-1_x86_64.ipk.zip)

解压以后安装里面的ipk

ggg-g, 2022-12-15:
> [kmod-r8168_5.4.188+8.051.02-1_x86_64.ipk.zip](https://github.com/istoreos/istoreos/files/10235512/kmod-r8168_5.4.188%2B8.051.02-1_x86_64.ipk.zip)
> 
> 解压以后安装里面的ipk

安装成功了

ggg-g, 2022-12-15:
> [kmod-r8168_5.4.188+8.051.02-1_x86_64.ipk.zip](https://github.com/istoreos/istoreos/files/10235512/kmod-r8168_5.4.188%2B8.051.02-1_x86_64.ipk.zip)
> 
> 解压以后安装里面的ipk

感谢大佬，我自己编译了好多天，一直不行。

jjm2473, 2022-12-15:
运行没问题吗

ggg-g, 2022-12-16:
> 运行没问题吗

大佬，安装完之后，路由器倒是正常运行。但我看不到运行的是哪个驱动程序

jjm2473, 2022-12-16:
执行 `ethtool -i eth0`

ggg-g, 2022-12-16:
> 执行 `ethtool -i eth0`

[image]
运行的还是原来的驱动

jjm2473, 2022-12-19:
更新固件吧，新固件已经删掉8169驱动了，换成了8168

ggg-g, 2022-12-19:
> 更新固件吧，新固件已经删掉8169驱动了，换成了8168

昨日已更新，目前运行良好

closed 2022-12-20 by jjm2473

JonXman, 2023-10-23:
> 更新固件吧，新固件已经删掉8169驱动了，换成了8168

大佬，我是istore OS的openwrt，他们固件没有支持r8168驱动，我安装你给的ipk后重启也还是r8169的驱动，怎么操作更换掉这个驱动呢？麻烦大佬你了

jjm2473, 2023-10-23:
@JonXman istoreos最新的固件已经可以在后台配置需要的驱动了。看看更新日志