istoreos/istoreos #479: 软件包升级问题

pctoolsx, 2022-11-24:
[image]
麻烦请问一下，这里的这些插件可以手动点升级吗？
因为我发现iStoreOS商店里的Alist是3.3.0，挂载阿里云有文件夹打不开，折腾了半天去alist GitHub手动下载了最新的3.5.1升级才行。

又看到这里提示有58个更新，不知道能不能升级？（上个版本的时候我试过手动更新，结果iStoreOS崩了）

pctoolsx, 2022-12-06:
好的，谢谢。

lxtdd, 2024-08-04:
看到这个帖子晚了，我试过几次nfs插件升级，每次点击升级后，第一插件版本没变，第二NFS管理界面就没了， 如何如何找不回，只能重刷。

hnzhaihui, 2024-09-17:
> 不要升级。我们已经在软件中心更新了 alist。 建议用沙箱模式，崩溃可以马上回退到上次好的版本。
> 
> 系统软件包不要升级，他们有依赖关系，升级之后，可能因为软件依赖问题，系统崩溃了！

115服务器崩溃后恢复，alist不能登录挂载的115网盘了，本人的安装的iStoreOS 22.03.6 2024032917 ，商店里的alist还是3.33.0，最新alist是3.37.3，已经解决了115网盘不能登录的问题，希望能尽快更新。

hnzhaihui, 2024-09-17:
> 不要升级。我们已经在软件中心更新了 alist。 建议用沙箱模式，崩溃可以马上回退到上次好的版本。
> 
> 系统软件包不要升级，他们有依赖关系，升级之后，可能因为软件依赖问题，系统崩溃了！

大神是否可以告知 在istroeos中商店里安装的软件的安装目录，我自己想办法去更改alist登录115的UA，就可以解决这个问题，谢谢！