istoreos/istoreos #483: 次新版在线升级到最新版，商店卡死。

lingyunzhiss, 2022-11-25:
This repository is a mirror of our main repo at https://git.openwrt.org/openwrt/openwrt.git
It is for reference only and is not active for checkins or for reporting issues.

All issues should be reported at: https://bugs.openwrt.org

Please do not open any NEW issue here - we will be closing the Github issues
capability in the near future.

We will continue to accept Pull Requests here as described in the repo description.

Close this now and click here: https://bugs.openwrt.org

Thankyou for your co-operation.
[image]

jjm2473, 2022-12-13:
估计是istore没有完整升级导致的，例如依赖的包没有升级

jjm2473, 2022-12-16:
打开终端，执行 `is-opkg install taskd luci-lib-taskd luci-app-store`，如果还有问题，就重启试试

closed 2023-01-04 by jjm2473