istoreos/istoreos #484: 关于“SysTools工具集”的几个问题。

BlackTouma, 2022-11-26:
SysTools工具集上手体验后的几个小问题。

1、关闭IPV6：能否增加开启IPV6的功能，能关闭也要能增加嘛。（笑

2、修复系统软件：运行后无法绑定易有云，重启后绑定就会失效，首页也一直显示未配置。

3、openssl speed -evp aes-256-gcm、openssl speed -evp chacha20-poly1305：运行后直接报错，无法测速。

BlackTouma, 2022-11-26:
@jannson 第二个问题，我发现是新版本固件的问题，我重置了系统，发现问题依然存在。绑定可以成功，重启后就显示离线。首页也一直显示未配置。

BlackTouma, 2022-11-30:
收到，辛苦了。

如果想重新开启IPV6的话，能否再增加一个快捷指令。还是在直接重新运行一遍关闭IPV6的指令就会重新开启呢？

nierainbow, 2023-07-04:
> SysTools工具集上手体验后的几个小问题。
> 
> 1、关闭IPV6：能否增加开启IPV6的功能，能关闭也要能增加嘛。（笑
> 
> 2、修复系统软件：运行后无法绑定易有云，重启后绑定就会失效，首页也一直显示未配置。
> 
> 3、openssl speed -evp aes-256-gcm、openssl speed -evp chacha20-poly1305：运行后直接报错，无法测速。

第一个问题我找到他的脚本，反向写了个脚本。启用ipv6
`#!/bin/sh

echo "uci settings"

uci -q batch <<-EOF
        del dhcp.lan.ra_flags
        set dhcp.lan.ra='server'
        set dhcp.lan.dhcpv6='server'
        
        set network.wan.ipv6='auto'
        del dhcp.wan.ra_flags
        set dhcp.wan.ra='server'
        
        set network.wan6.auto='1'
        
        delete dhcp.@dnsmasq[0].filter_aaaa
        
        commit network
        commit dhcp
EOF

echo "reload_config"
reload_config
echo -n "wait for device online"
for i in `seq 10 -1 1`; do
        echo -n "."
        sleep 1
done
echo ""
echo "done"
`