istoreos/istoreos #487: 软件包更新 curl 出现Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or directory (needed by /usr/bin/curl)

harveyhonorg, 2022-11-26:
设备 FriendlyElec NanoPi R5S
架构 ARMv8 Processor rev 0
目标平台 rockchip/rk3568
固件版本 | iStoreOS 21.02.3 2022112412 / LuCI openwrt-21.02 branch git-22.304.65299-04257f6

软件包更新 curl软件包后，在安装其他软件包出现错误

Upgrading ip full

Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or directory (needed by /usr/bin/curl)
Collected errors:
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_cortex-a53/base/ip-full_5.11.0-3_aarch64_cortex-a53.ipk, curl returned 127.
 * opkg_install_pkg: Failed to download ip-full. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package ip-full.

Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or directory (needed by /usr/bin/curl)
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or directory (needed by /usr/bin/curl)
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or directory (needed by /usr/bin/curl)
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or directory (needed by /usr/bin/curl)
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or directory (needed by /usr/bin/curl)
Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or directory (needed by /usr/bin/curl)
Collected errors:
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/targets/rockchip/rk3568/packages/Packages.gz, curl returned 127.
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_cortex-a53/base/Packages.gz, curl returned 127.
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_cortex-a53/luci/Packages.gz, curl returned 127.
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_cortex-a53/packages/Packages.gz, curl returned 127.
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_cortex-a53/routing/Packages.gz, curl returned 127.
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_cortex-a53/telephony/Packages.gz, curl returned 127.

@£££这个是目前找到的解救方案，opkg install libwolfssl24 这个软件包目前没找到下载路径，系统解决下冲突

https://github.com/vernesong/OpenClash/issues/905

alanljj, 2022-11-29:
我也碰到同样的问题

harveyhonorg, 2022-11-29:
https://doc.linkease.com/zh/guide/istoreos/question.html#%E6%8F%92%E4%BB%B6%E5%AE%89%E8%A3%85%E5%A4%B1%E8%B4%A5-%E6%8F%90%E7%A4%BA-curl-%E9%94%99%E8%AF%AF
你用这个试试看行不行，我是重新安装了，不在更新CURL这个软件包，目前没问题

| 发件人 | ***@***.***> |
| 发送日期 | 2022年11月29日 20:34 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***> ,
***@***.***> |
| 主题 | Re: [istoreos/istoreos] 软件包更新 curl 出现Error loading shared library libwolfssl.so.5.5.1.99a5b54a: No such file or directory (needed by /usr/bin/curl) (Issue #487) |

我也碰到同样的问题

nevertoday, 2022-12-07:
我也遇到这个问题了哈， 现在检查更新
[image]

nevertoday, 2022-12-07:
root@iStoreOS:~# opkg update
Package curl version 7.83.1-1 has no valid architecture, ignoring.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/targets/rockchip/armv8/packages/Packages.gz
*** Failed to download the package list from https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/targets/rockchip/armv8/packages/Packages.gz

Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/base/Packages.gz
*** Failed to download the package list from https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/base/Packages.gz

Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/luci/Packages.gz
*** Failed to download the package list from https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/luci/Packages.gz

Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/packages/Packages.gz
*** Failed to download the package list from https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/packages/Packages.gz

Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/routing/Packages.gz
*** Failed to download the package list from https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/routing/Packages.gz

Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/telephony/Packages.gz
*** Failed to download the package list from https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/telephony/Packages.gz

Collected errors:
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/targets/rockchip/armv8/packages/Packages.gz, curl returned 255.
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/base/Packages.gz, curl returned 255.
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/luci/Packages.gz, curl returned 255.
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/packages/Packages.gz, curl returned 255.
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/routing/Packages.gz, curl returned 255.
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.3/packages/aarch64_generic/telephony/Packages.gz, curl returned 255.

0x1af2aec8f957, 2023-02-10:
> 请问， 有没有傻瓜一点点的办法， 可以恢复哈， 我现在什么软件包都不能安装。

@nevertoday 
```bash
opkg update
opkg install libcurl
```

可以修复这个问题。

wulin5500, 2023-02-28:
> > 请问， 有没有傻瓜一点点的办法， 可以恢复哈， 我现在什么软件包都不能安装。
> 
> @nevertoday
> 
> ```shell
> opkg update
> opkg install libcurl
> ```
> 
> 可以修复这个问题。

666

mustundead, 2023-03-10:
> > 请问， 有没有傻瓜一点点的办法， 可以恢复哈， 我现在什么软件包都不能安装。
> 
> @nevertoday
> 
> ```shell
> opkg update
> opkg install libcurl
> ```
> 
> 可以修复这个问题。

哭了，卡我了好几天

PolymorphismException, 2023-05-25:
> > 请问， 有没有傻瓜一点点的办法， 可以恢复哈， 我现在什么软件包都不能安装。
> 
> @nevertoday
> 
> ```shell
> opkg update
> opkg install libcurl
> ```
> 
> 可以修复这个问题。

牛牛牛

eRain03, 2023-07-09:
> > 请问， 有没有傻瓜一点点的办法， 可以恢复哈， 我现在什么软件包都不能安装。
> 
> @nevertoday
> 
> ```shell
> opkg update
> opkg install libcurl
> ```
> 
> 可以修复这个问题。

牛

ByteSecurity, 2023-07-15:
如果出现 
`Error relocating /usr/lib/libcurl.so.4: nghttp2_option_set_no_rfc9113_leading_and_trailing_ws_validation: symbol not found`

可以用 
` opkg install libnghttp2`

好运~

app911, 2024-05-16:
> > 请问， 有没有傻瓜一点点的办法， 可以恢复哈， 我现在什么软件包都不能安装。
> 
> @nevertoday
> 
> ```shell
> opkg update
> opkg install libcurl
> ```
> 
> 可以修复这个问题。

谢谢~