istoreos/istoreos #490: Frpc 报 RPCError  版本

fytiaopi, 2022-11-27:
设备：FriendlyElec NanoPi R2S 友善R2S 
系统环境：r5s-update-0610.img On Emmc存储
固件版本：iStoreOS 21.02.3 2022112309
内核版本：5.4.188
报错插件：frpc v1.0.0
安装方式：iStore [v0.1.10-4]软件商店自动安装
错误信息：
RPCError
RPC call to uci/get failed with ubus code 4: 未找到资源
  at ClassConstructor.handleCallReply (http://192.168.1.3/luci-static/resources/rpc.js?v=git-22.320.13537-19b3268:15:3)

sakura0224, 2022-12-27:
同样的问题
[image]

fytiaopi, 2022-12-27:
我已经放弃了 刷了另外的版本自己装了

sakura0224, 2022-12-27:
> 我已经放弃了 刷了另外的版本自己装了

请问你用的是什么版本？

fytiaopi, 2022-12-27:
> 

https://github.com/stupidloud/nanopi-openwrt
我用了个没有任何插件的slim  openwrt 自己装的frpc

coolzhi, 2023-01-20:
frpc还是老的版本好用,新的luci里面http端口都没有设置的地方(可能是我不会用，反正不方便),现在手动安装的esir网盘里的0.34的版本用的luci也用的老的，我永起来很方便

jjm2473, 2023-02-23:
重复 https://github.com/istoreos/istoreos/issues/419

closed 2023-02-23 by jjm2473