istoreos/istoreos #493: 校园网的ipv6地址获取问题

luyi1680, 2022-11-28:
校园网有ipv6，如果网线直连电脑，电脑pppoe拨号是可以获得ipv4和ipv6地址的。但是用装有istoreOS的软路由拨号就无法获取ipv6地址，请问如何解决。

My-By, 2022-12-01:
能给我看看你的设置吗，有图的话我好帮你看看能不能解决。

luyi1680, 2022-12-01:
> 能给我看看你的设置吗，有图的话我好帮你看看能不能解决。

感谢回复，我开始想按照https://post.smzdm.com/p/ad9qkz9k/和https://sustech.online/service/network/ipv6/两篇文章里的设置方法进行设置，发现其有些界面的选项与istoreos里的不一致，下面我的设置。

WAN-高级设置：
[image]
[image]
[image]
[image]
[image]

WAN6：
[image]
[image]
[image]
[image]
[image]

LAN：
[image]
[image]
[image]
[image]
[image]
[image]
[image]

DHCP/DNS：
[image]

防火墙：
[image]

My-By, 2022-12-01:
奇怪，讲道理你这个设置应该是可以分配到ipv6的啊，

My-By, 2022-12-01:
你把网络接口的页面截图我看看，

luyi1680, 2022-12-01:
[image]

My-By, 2022-12-01:
你这个已经分配到ipv6了，你看lan口的ipv6有显示了。

luyi1680, 2022-12-01:
这个不是应该得有2403开头的IP才行嘛

luyi1680, 2022-12-01:
[image]
首页这里也不显示ipv6

My-By, 2022-12-01:
右键电脑右下角的网络打开网络和in的设置-更改配适器选项-右键你的网卡-状态，看看有没有ipv6，界面里是不会同时显示v6和v4的，有v4会优先跑v4，没有v4才跑v6，那个显示的是你现在再跑的网络。

luyi1680, 2022-12-01:
也是没有
[image]

My-By, 2022-12-01:
所以你是有v6了，只是软路由界面没显示而已，那个没办法，强迫症会觉得不舒服，我以前也是这样，

luyi1680, 2022-12-01:
我再研究研究，谢谢你

jjm2473, 2023-01-10:
只有2开头的ipv6地址才是有效的公网地址，如果没有分配到，可能要找其他人问问是不是有ipv6地址，以及如何获取（PPPOE还是DHCPv6），也可以试试电脑拨号以后能不能获取到ipv6地址

My-By, 2023-01-18:
你的问题解决了吗 我大概知道你的问题了，如果没解决，回复我一下，我跟你说怎么解决。

closed 2023-03-13 by jjm2473

passwii, 2023-03-17:
> 
Hi，我遇到和他同样的问题，PPPOE的V6有显示，但是无法通过IPV6联网测试； 已经关闭DHCP/DNS里面的禁止IPV6解析。 请问你如何解决这个问题？

luyi1680, 2023-03-17:
> > 
> 
> Hi，我遇到和他同样的问题，PPPOE的V6有显示，但是无法通过IPV6联网测试； 已经关闭DHCP/DNS里面的禁止IPV6解析。 请问你如何解决这个问题？

我参考这个网站里的内容进行设置，就可以了
https://sustech.online/service/network/ipv6/