istoreos/istoreos #498: istoreos的Home Assistant

Soul8310, 2022-11-30:
istoreos的Home Assistant里打开了高级模式，再配配置里少了加载项，缺少了很多插件

Soul8310, 2022-11-30:
> 方便截图么？免得我可能不好理解。
>
>
> 从QQ邮箱发来的超大附件
>
> 1.png (1.95M, 无限期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=0f3362372b84d9c91296341414640b195352510e560101021804510054490d5754004f56055d0b1b505556035705015453030356326139071b430c503259&t=exs_ftn_download&code=53b72d96
>
> 2.png (2.19M, 无限期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=5e386238d0d6eec7439d341b14380017500e5159045e5659495d565c0615060a075d4f59510c0015055e07595601505a050c5a5b323d320a4a480c5f3205&t=exs_ftn_download&code=d8b82828
>
> 3.png (1.91M, 无限期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=023664320700d1c71f9332111e620217080205005c04010a150353510f4f04010f06490a5d070715005757540d5509090e055d513867300b16460a55385f&t=exs_ftn_download&code=86d28b08
>
> 4.png (1.95M, 无限期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=0c383932cdf8d0cb119d6f111630021b06080f01085302501b090f03011d04040f59140b06015319545e5c575105080c52595a013035300018485755300d&t=exs_ftn_download&code=68920004

jjm2473, 2023-03-16:
加载项是supervisor版本的功能，实际就是让HA可以安装更多docker容器。supervisor只有x86镜像，所以我们不可能支持。

closed 2023-03-16 by jjm2473

murazi, 2023-04-25:
> 加载项是supervisor版本的功能，实际就是让HA可以安装更多docker容器。supervisor只有x86镜像，所以我们不可能支持。

我就是x86下的istoreOS，试了ISOS默认提供的几个HA镜像，都不带HA官方商店
[image]

目前用的是稳定版

jjm2473, 2023-05-27:
都说了不是同一个镜像，supervisor不是通用的镜像，你要安装的话可以自己看看官网这么用docker安装supervisor

tazyong, 2023-06-02:
iStore 商店的 HA 不好玩，直接装 hassio 吧 #815

murazi, 2023-06-05:
> iStore 商店的 HA 不好玩，直接装 hassio 吧 #815

看了官方文档,docker下似乎就不支持HASSIO

tazyong, 2023-06-05:
> > iStore 商店的 HA 不好玩，直接装 hassio 吧 #815
> 
> 看了官方文档,docker下似乎就不支持HASSIO

我都装上玩了几天了，你跟我说不支持？issue 你也没看，截图都放出来了...

jjm2473, 2023-06-15:
hassio是只支持x86的，但是这个插件是要给所有平台使用的，所以用的是通用的镜像。前面已经说过了