istoreos/istoreos #517: 比较严重的bug，希望能修复下

makeye, 2022-12-11:
This repository is a mirror of our main repo at https://git.openwrt.org/openwrt/openwrt.git
It is for reference only and is not active for checkins or for reporting issues.

All issues should be reported at: https://bugs.openwrt.org

Please do not open any NEW issue here - we will be closing the Github issues
capability in the near future.

We will continue to accept Pull Requests here as described in the repo description.

Close this now and click here: https://bugs.openwrt.org

Thankyou for your co-operation.

我正常的使用openclash和smartdns插件也能够正常上网。但是时不时就会抽风，主页网络状态就会出现dns错误或软件源错误，不管他过老长一段时间，他又自己好了。从16814版本就开始出现问题。机子是ars2

makeye, 2022-12-11:
出现问题之后就无法联网了

makeye, 2022-12-11:
麻烦问下需要哪些日志？

ccatmao, 2024-01-12:
同样问题