istoreos/istoreos #518: r5s 最新固件 跑不满千兆

handsomebroB, 2022-12-12:
试了很多次，老的固件和新的固件都无法跑到千兆，其他的固件是可以跑满千兆的，是什么原因呢？

handsomebroB, 2022-12-12:
用您提到的方式内网测速没有问题，能跑满2.5g
我遇到的问题是外网测速，目前来看，我试用过官方固件，lean固件和supes的固件,这三个固件才测速时基本能跑到950M，网络不稳定时也能跑到850M左右，（没有再截图了o(*￣︶￣*)o，当然他们也都有其他的bug或者不方便的地方。）

- 我刚刚也对比了您固件的x86版本，我用hyperv虚拟机安装的，这是在windows下用speedtest测速的情况：
[image]
大概是能跑到850M左右的。
- 下边是换到r5s最新固件后用windows下speedtest测速的情况：
[image]
[image]
无法跑到850M。

- 下边是用路由器上的测速工具测到的情况：
[image]
[image]
[image]
同样存在无法跑到850M的情况，也会出现不稳定的状况。

我家是成都电信1000m的宽带，我测试的时候是晚上八点左右，可能是高峰期。没办法到950m左右可能是网络问题。但之前网络通畅的时候也遇到了同样的问题，官方固件可以跑满950m的时候，istoreos也只能跑到500m左右。
不知道是bug还是什么其他原因。

此外还有一个小问题。opkg update里有一条无法连接，试了所有的源都是一样的，好像x86没有这个问题：
[image]
辛苦大神了！！^_^

TeddyXiong79, 2022-12-23:
和我的情况一模一样。。。。

TeddyXiong79, 2022-12-25:
没有人跟一下吗？ 谢谢大佬哦

TeddyXiong79, 2022-12-25:
再次Update，换Friendly原生固件就没有问题。

Mess-R, 2023-01-12:
一样的问题。换了固件就不会。
[image]

handsomebroB, 2023-02-09:
最新的固件已经修复了这个问题了，我已经能够跑满带宽了，666

f380329400, 2023-02-16:
你这个问题是怎么解决的呢，我用2月3号的那个版本还是没解决这个问题呢。

TeddyXiong79, 2023-02-21:
没解决啊。。。我的还是一样。。

TeddyXiong79, 2023-03-06:
大家看看这个问题有解决吗？？？？感觉是Driver的问题。。。。。自己也不会单独升级网卡的Driver。。。。请管理员关注一下哦。

Mess-R, 2023-03-08:
> 大家看看这个问题有解决吗？？？？感觉是Driver的问题。。。。。自己也不会单独升级网卡的Driver。。。。请管理员关注一下哦。

我换固件。

xzkyo, 2023-07-03:
R4s用你的最新版本遇到同样的问题，跑不满千兆，只能600多，而且把一些该卸载的软件都卸了，还是一样。换到普通路由器能到900

TeddyXiong79, 2023-07-03:
已经解决了，升级到当前最新的固件就好。

virtually321, 2023-10-30:
并没有，我认为是网卡驱动的问题。以前千兆测速能到930的速度，现在到头就到830

TeddyXiong79, 2023-10-30:
我升级2023年6月份的那个固件，问题已经解决了。目前最新的固件我没测试。