istoreos/istoreos #522: 加上vitio网卡驱动

GavinIves, 2022-12-13:
如题 希望可以加一个 这个效率很高

jjm2473, 2022-12-15:
openwrt里面没找到这个驱动，如果你在哪个opewnrt固件看到有，把固件的仓库发出来

jjm2473, 2022-12-15:
我看内核已经集成了VIRTIO_NET，VIRTIO_BLK啊，你不是在x86平台吗  https://github.com/istoreos/istoreos/blob/6c0a7f4ee29d0db4e964ee79dfaa47ea6d7369e1/target/linux/x86/64/config-5.4#L421-L438

GavinIves, 2022-12-17:
> 我看内核已经集成了VIRTIO_NET，VIRTIO_BLK啊，你不是在x86平台吗
> 
> https://github.com/istoreos/istoreos/blob/6c0a7f4ee29d0db4e964ee79dfaa47ea6d7369e1/target/linux/x86/64/config-5.4#L421-L428
我是在PVE平台搭建的 我通过命令查看无法识别vitio驱动 
[image]
并且测速无法超过100mb 怀疑协商速率只有100mb
[image]

jjm2473, 2022-12-19:
`ethtool -i eth0`看看

GavinIves, 2022-12-19:
[image]
[image]
这么看他貌似识别到了  但它的协商速度都不对劲 我的宿主机网卡是x540-at2

GavinIves, 2022-12-19:
[image]
这是我另一台虚拟机 也是vitio网卡 貌似他的输出和istore一样 但它的速度可以达标

GavinIves, 2022-12-19:
[image]
这是VMware vmxnet3网卡在首页的展示 
[image]
这是vitio网卡

jjm2473, 2022-12-25:
我用kvm部署了istoreos （ https://github.com/istoreos/istoreos-kvm ），使用的是virtio_net，测试没有问题，速度可以超过 10G

GavinIves, 2022-12-25:
我用的是x86版本的 img直接导入pve变成硬盘的 可以试一下这种方式的结果吗

GavinIves, 2022-12-25:
我有一点奇怪的是 从istoreos第一版开始 我就一直用到现在 都是下载的x86版本的img 直接导入进入pve的 所有版本都有问题 这应该不是我的个例 但是却没有别人的issue 是我的使用方法有问题吗

jjm2473, 2023-01-03:
估计是pve的问题，或者cpu分得不够，既然qemu-kvm部署的速度没问题，那就说明client的驱动没问题了

closed 2023-01-03 by jjm2473

jjm2473, 2023-01-04:
实测j4125单核，pve里面跑istoreos，虚拟网卡的速度超过11Gbps，所以不可能是驱动问题。

GavinIves, 2023-01-04:
可以分享一下pve中虚拟机的配置吗 我好检查一下 万分感谢

GavinIves, 2023-01-04:
最新版本确实支持了 我升级之后ok了 很奇怪

jjm2473, 2023-01-08:
我怀疑是你的网线问题，导致物理网卡协商到百兆了

GavinIves, 2023-01-08:
相信我 之前我把所有的可能性都排除了 期间也没有换网线 在上一个版本确实只能百兆 最新版本就可以了

GavinIves, 2023-01-08:
我是万兆七类网线 不过现在问题已经解决了 万分感谢您的帮助