istoreos/istoreos #524: R2S无法使用wireguard，添加相关接口就崩溃

666dechaoge, 2022-12-14:
[42722.701657] br-lan: port 1(eth1) entered disabled state
[42722.704800] r8152 4-1:1.0 eth1: carrier off
[42731.630706] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[42731.631040] r8152 4-1:1.0 eth1: carrier on
[42731.631712] br-lan: port 1(eth1) entered blocking state
[42731.631729] br-lan: port 1(eth1) entered forwarding state
[42915.234669] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[42922.363420] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[42925.633712] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000018
[42925.634518] Mem abort info:
[42925.634782]   ESR = 0x96000006
[42925.635061]   EC = 0x25: DABT (current EL), IL = 32 bits
[42925.635537]   SET = 0, FnV = 0
[42925.635813]   EA = 0, S1PTW = 0
[42925.636092] Data abort info:
[42925.636348]   ISV = 0, ISS = 0x00000006
[42925.636687]   CM = 0, WnR = 0
[42925.636958] user pgtable: 4k pages, 48-bit VAs, pgdp=000000001b521000
[42925.638128] [0000000000000018] pgd=000000001887a003, pud=000000001ac4c003, pmd=0000000000000000
[42925.638934] Internal error: Oops: 96000006 [#1] PREEMPT SMP
[42925.639438] Modules linked in: oaf xt_FULLCONENAT rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async mt76x0u mt76x0_common iwlmvm iwldvm ath10k_pci ath10k_core ath wireguard rtlwifi rtl8812au rndis_host pppox ppp_mppe ppp_generic pl2303 mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_common mt7615_common mt76_usb mt76_connac_lib mt76 mmc_spi mac80211 libchacha20poly1305 libblake2s iwlwifi ipt_REJECT cp210x chacha_neon ch341 cfg80211 cdc_ether ax88179_178a asix zstd xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_bpf xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY via_velocity via_rhine usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200
[42925.639605]  ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_fsm ts_bm slhc sky2 skge sis900 sis190 sfp sfc_falcon sfc r8169 r8152 r8125 r6040 poly1305_neon pcnet32 of_mmc_spi ntfs3 niu nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount ne2k_pci mlx5_core mlx4_en mlx4_core mdio_i2c mdio_gpio mdio_bitbang macvlan lzo libcurve25519_generic libchacha libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables igc iavf forcedeth exfat ethoc et131x e100 crc7 crc_ccitt compat br_netfilter bnx2x bnx2 be2net atl2 atl1e atl1c atl1 asn1_decoder alx 8390 8139too 8139cp fuse act_connmark
[42925.647311]  sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs sg pwm_fan drivetemp spi_ks8995 siit xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink st sr_mod cdrom rtl8366s rtl8366rb rtl8306 ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip17xx b53_mdio b53_common nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 nfsd nfs msdos bonding 3c59x vmxnet3 ixgbevf ixgbe igb i2c_algo_bit e1000 ifb dummy mdio ip_tunnel hfcpci hfcmulti rpcsec_gss_krb5 auth_rpcgss oid_registry dm_raid veth tun nbd mISDN_dsp l1oip mISDN_core xfrm_user xfrm_ipcomp af_key
[42925.655048]  xfrm_algo vfat fat udf crc_itu_t lockd sunrpc grace minix isofs hfsplus hfs cramfs cifs autofs4 rxrpc dns_resolver dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax br2684 atm aoe multipath fscache raid456 async_raid6_recov async_pq async_xor async_memcpy async_tx raid10 raid1 raid0 linear md_mod nls_utf8 nls_cp936 nls_cp437 natsemi vxlan udp_tunnel ip6_udp_tunnel sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg pcbc md5 md4 kpp hmac ghash_generic gf128mul gcm fcrypt echainiv ecb des_generic libdes deflate cts ctr cmac ccm cbc authenc arc4 crypto_acompress fb font drm drm_panel_orientation_quirks uhci_hcd rtl8367b swconfig rtl8366_smi sata_via sata_sil24 sata_sil ata_piix pata_pdc202xx_old sata_nv sata_mv pata_artop ahci libahci libata fsl_mph_dr_of ehci_fsl gpio_button_hotplug xfs reiserfs jfs btrfs xor zstd_decompress zstd_compress xor_neon raid6_pq lzo_decompress lzo_compress button_hotplug tg3 b44 ssb broadcom bcm_phy_lib bcm84881
[42925.670341] CPU: 2 PID: 12331 Comm: modprobe Not tainted 5.4.188 #0
[42925.670898] Hardware name: FriendlyElec NanoPi R2S (DT)
[42925.671374] pstate: 00000005 (nzcv daif -PAN -UAO)
[42925.671823] pc : string_nocheck+0x10/0x80
[42925.672187] lr : string+0x54/0x68
[42925.672488] sp : ffff800012a93b50
[42925.672791] x29: ffff800012a93b50 x28: 0000000000000d5d 
[42925.673264] x27: ffff00002c0492a3 x26: 00000000ffffffd8 
[42925.673739] x25: ffff8000108b1bf0 x24: ffff800012a93ce0 
[42925.674219] x23: 0000000000000002 x22: ffff00002c04a000 
[42925.674697] x21: ffff0a00ffffff04 x20: 0100000000000018 
[42925.675173] x19: ffff00002c04a000 x18: 0000000000000000 
[42925.675651] x17: 0000000000000000 x16: 0000000000000000 
[42925.676129] x15: 0000000000000000 x14: 0000000000000000 
[42925.676606] x13: 0000000000000000 x12: 0000000000000008 
[42925.677083] x11: 0101010101010101 x10: ffff800012a93ce0 
[42925.677559] x9 : 00000000ffffffd0 x8 : ffff00002c049000 
[42925.678035] x7 : ffff800012a93ce0 x6 : ffffffffffffffff 
[42925.678508] x5 : 00000000ffffffd0 x4 : 0000000000000000 
[42925.678983] x3 : ffff0a00ffffff04 x2 : 0100000000000018 
[42925.679465] x1 : ffff00002c04a000 x0 : ffff00002c0492a3 
[42925.679947] Call trace:
[42925.680179]  string_nocheck+0x10/0x80
[42925.680512]  string+0x54/0x68
[42925.680779]  vsnprintf+0x394/0x748
[42925.681092]  seq_vprintf+0x5c/0x80
[42925.681401]  seq_printf+0x44/0x50
[42925.681709]  m_show+0xa4/0x210
[42925.681986]  seq_read+0x390/0x500
[42925.682286]  proc_reg_read+0x70/0xb8
[42925.682614]  __vfs_read+0x18/0x40
[42925.682912]  vfs_read+0xc8/0x158
[42925.683203]  ksys_read+0x4c/0xc8
[42925.683494]  __arm64_sys_read+0x18/0x20
[42925.683846]  el0_svc_common.constprop.2+0x7c/0x110
[42925.684276]  el0_svc_handler+0x20/0x80
[42925.684615]  el0_svc+0x8/0x600
[42925.684902] Code: a9bf7bfd 9370fc66 910003fd 340002e6 (39400045) 
[42925.685447] ---[ end trace c5d0efdce594056d ]---

jjm2473, 2022-12-15:
下个版本会集成wireguard到固件里，应该就可以解决这个问题了

jjm2473, 2022-12-19:
最新固件已经集成wireguard驱动，看看是不是解决了

666dechaoge, 2022-12-19:
> 最新固件已经集成wireguard驱动，看看是不是解决了

没有公网IP，使用frpc穿透出去的wireguard，测试可以正常使用，非常感谢

closed 2022-12-19 by 666dechaoge