istoreos/istoreos #525: iStoreOS核显驱动问题 N5105

KeplerSSSSS, 2022-12-14:
麻烦大佬给iStoreOS添加N5105的核显驱动  用以支持jellyfin硬件解码  保持CPU运行稳定

参考文献 https://post.smzdm.com/p/amm500lv/
                https://post.smzdm.com/p/aen5k88z/

jjm2473, 2022-12-15:
目前istoreos的linux内核是5.4，不支持n5105的硬解

KeplerSSSSS, 2022-12-16:
> 

得升级到什么版本的内核可以支持硬解呢?

iStoreOS有没有这方面的计划?

不能用上n5105的核显,相当于把这个U给阉割了啊

jjm2473, 2022-12-16:
好像要5.18以上才支持，目前上游的openwrt都还没升级到这个版本

jian263994241, 2022-12-21:
> 
5.4 5.18 那个版本高?   这版本号怎么看

jjm2473, 2022-12-25:
> 5.4 5.18 那个版本高? 这版本号怎么看
5.18高。`uname -a`可以看到内核版本

shen-lan, 2023-01-10:
都买N5105了，肯定是虚拟机跑的 istore吧。
istore 跑路由科学功能， 再安装个linux， 跑 docker服务。jellyfin， emby等服务都在这里的docker

a912327347, 2023-01-10:
现在最新的版本内核已经到5.18了  可以驱动核显了吗？

jjm2473, 2023-01-12:
> 现在最新的版本内核已经到5.18了 可以驱动核显了吗？

openwrt的master分支都还没更新啊，更别提release了。
如果要用最新内核，建议使用其他固件源码，例如lede，加上istoreos相关的补丁就行了，之前有人做过这样的事

jjm2473, 2023-04-18:
测试固件支持了 https://github.com/istoreos/istoreos/issues/716

hongxuWei, 2023-04-18:
太棒了，期待能用上

jjm2473, 2023-10-31:
最新的正式版本固件已经移植了i915-oot驱动，支持所有intel的gpu。

closed 2023-10-31 by jjm2473

shen-lan, 2023-12-29:
@jjm2473  
你好。我是N5105物理机安装的，现在是最新版固件，但是试了, 无论是emby, 还是jellyfin都不能硬解（转码设置里都设置了硬解， /dev/dri/下两个文件是有的）。似乎还是不能硬解。
设备： N5105
版本： [Powered by LuCI istoreos-22.03 branch (git-23.342.39264-37c783a)](https://github.com/openwrt/luci)
问题： 
emby播放一个mkv视频， 使用的是cpu解码，后台看到cpu占用99%， 温度70多度。我又换了个h264或者其他常见编码的视频测试，公网下播放，故意调低分辨率，也是一样cpu直接被拉满。

-------------------------------------------
其他issue里问题指导尝试
```
root@iStoreOS:~# lspci -tv
-[0000:00]-+-00.0  Intel Corporation Device 4e24
           +-02.0  Intel Corporation JasperLake [UHD Graphics]
           +-14.0  Intel Corporation Device 4ded
           +-14.2  Intel Corporation Device 4def
           +-14.5  Intel Corporation Device 4df8
           +-15.0  Intel Corporation Serial IO I2C Host Controller
           +-15.2  Intel Corporation Device 4dea
           +-16.0  Intel Corporation Management Engine Interface
           +-17.0  Intel Corporation Device 4dd3
           +-1c.0-[01]----00.0  Realtek Semiconductor Co., Ltd. Device 5765
           +-1c.4-[02]----00.0  Intel Corporation Device 125c
           +-1c.5-[03]----00.0  Intel Corporation Device 125c
           +-1c.6-[04]----00.0  Intel Corporation Device 125c
           +-1c.7-[05]----00.0  Intel Corporation Device 125c
           +-1f.0  Intel Corporation Device 4d87
           +-1f.3  Intel Corporation Jasper Lake HD Audio
           +-1f.4  Intel Corporation Jasper Lake SMBus
           \-1f.5  Intel Corporation Jasper Lake SPI Controller
```

```
root@iStoreOS:~# dmesg | grep i915
[    8.106324] disable_kmod.group i915-oot
[    8.716431] i915 0000:00:02.0: [drm] Incompatible option enable_guc=3 - GuC submission is N/A
[    8.731295] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/icl_dmc_ver1_09.bin (v1.9)
[    8.767158] i915 0000:00:02.0: [drm] GuC firmware i915/ehl_guc_33.0.4.bin version 33.0 submission:disabled
[    8.767163] i915 0000:00:02.0: [drm] HuC firmware i915/ehl_huc_9.0.0.bin version 9.0 authenticated:yes
[    8.778039] [drm] Initialized i915 1.6.0 20200917 for 0000:00:02.0 on minor 0
[    8.780175] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    8.780285] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
```

jjm2473, 2024-01-02:
@shen-lan 你的jellyfin配置了硬解？你的jellyfin这么部署的，是用istore里面的插件还是手动的？不管固件，你以前在x86上成功开启jellyfin硬解吗？

shen-lan, 2024-01-07:
现在用istore商店里的可以了。docker按官网部署的都不行

jjm2473, 2024-01-07:
@shen-lan istore的插件部署x86的jellyfin就是使用官方的docker镜像，我怀疑你手动部署的时候没把/dev/dri映射到容器里。

shen-lan, 2024-01-07:
顺便问个其他的问题，istore运行一段时间后，局域网smb传输文件，速度只有几m/s(是的，你没听错，就是这么离谱 😅😅😅)，最高不超过20m/s。重启机器就正常了，能达到将近100m/s。这个情况有见过吗

jjm2473, 2024-01-08:
@shen-lan emby后台配置硬解了吗？至少应该打开低电压编码，解码格式勾选除AV1以外所有格式。

jjm2473, 2024-01-08:
@shen-lan 还有，之前你部署的jellyfin，版本是10.8以上的吗？

shen-lan, 2024-01-08:
@jjm2473  肯定是配置了。

shen-lan, 2024-01-08:
@jjm2473  最新的dokcer镜像。

jjm2473, 2024-01-08:
@shen-lan 不要看是不是最新，如果你使用了国内的镜像，例如阿里云的，根本不会更新tag。要进后台的dashboard，看版本号

shen-lan, 2024-01-08:
科大的镜像源，具体版本还真没看....
算了，jellyfin已经可以了。不浪费精力了...谢谢您的热心解答

jjm2473, 2024-01-08:
@shen-lan 你手动部署的时候有加“--privileged”标志吗

shen-lan, 2024-01-08:
特权模式是吧   有的。

jjm2473, 2024-01-08:
@shen-lan 科大的镜像源就是阿里云的

shen-lan, 2024-01-08:
.....好吧  可能 是版本的问题。谢谢热心解答