istoreos/istoreos #526: R66s 无法软重启！重启其实是关机。

xpn282, 2022-12-15:
这个bug很久了，一直没有修复。每次重启都要拔电源才行

jjm2473, 2023-01-10:
r68s可以正常重启，难道使用sd卡的机型就无法重启？

jjm2473, 2023-01-10:
有没有试过换个sd卡

jjm2473, 2023-01-10:
找人测试了r66s最新固件，软重启没问题，建议用最新固件重新刷tf卡

xpn282, 2023-01-11:
换卡也不行。我不是个例，r66s群里一大堆无法软重启的，但也有几个人能软重启……玄学

jjm2473, 2023-01-12:
之前老版本固件使用的uboot是openwrt提供的，可能这个版本有问题，后来换成了rockchip提供的uboot。但是如果只是升级固件，不会升级uboot，所以用最新固件重新写卡试试，你也可以跟其他人说下这个，看看能不能解决问题。
如果上面的方法无效，可能就不是软件的问题了，可能是tf卡或者硬件的问题，问下有同样问题的其他人硬件版本是什么（电路板有打印），以及使用的tf卡的品牌型号

xpn282, 2023-01-12:
我换了一张很老的tf 8g卡刷最新的固件，可以重启了。。。
但是之前的那张卡就算重新刷最新的固件也不能重启，同样的卡，用别的固件是没有这个问题的，那只能说iStoreOS挑卡了，难道是内核的原因，毕竟4.19的内核比较老。

jjm2473, 2023-01-12:
跟内核的关系不大，可能是uboot的问题，所以说重新写卡也许能解决，避免使用旧的uboot。

xpn282, 2023-01-12:
都是重刷固件，有的卡行，有的不行。不行的卡刷其他固件是可以重启，只能说iStoreOS挑卡了，希望iStoreOS能改进咯，更好的兼容tf卡

jjm2473, 2023-01-12:
根据其他openwrt的补丁，可能是要给sd卡降速，重启才不会有问题，我还没试过。参考这些补丁：
https://github.com/coolsnowwolf/lede/blob/6fc2d00f257c988ccb27e618317a0c83f496f103/target/linux/rockchip/patches-5.4/107-nanopi-r4s-sd-signalling.patch
https://github.com/coolsnowwolf/lede/blob/6fc2d00f257c988ccb27e618317a0c83f496f103/target/linux/rockchip/patches-5.4/105-mmc-core-set-initial-signal-voltage-on-power-off.patch
https://github.com/openwrt/openwrt/blob/2b583ab8a74a6bd41f1c9629b3b66ba561d6fceb/target/linux/rockchip/patches-5.10/105-nanopi-r4s-sd-signalling.patch

xpn282, 2023-01-12:
太深奥的我也不懂…为了重启，我现在换用了10多年前的卡了……也是无语

jjm2473, 2023-01-13:
@xpn282  用之前不能重启的sd卡试试这个固件 https://we.tl/t-t8AwDrL3Pk

xpn282, 2023-01-13:
试了，还是不能。。不折腾了，继续用老的卡