istoreos/istoreos #534: https 进入， 但是为什么总很多服务，自动变成http

nevertoday, 2022-12-23:
我已经证书和相关ddns 之类的配置，都处理成功， 且能正确访问！
但是目前让我困惑的是，很多服务，比如home Assiant 还有 adGuard Home 之类的服务，只要我点击自定义的域名访问，然后点击具体的业务，都不会自动变成https 链接访问， 而是http访问；
从istoreOS后台中也不曾有openwrt 的重定向80到443 端口， 
我的问题是：
如何实现默认 https 访问我的docker 之类的端口服务， 而不是总默认是http访问， 

我的uhttpd 配置如下： 
[image]

lmlx66, 2023-01-08:
跑一个nginx监听端口强制反代https，这并不是bug问题

jjm2473, 2023-01-10:
这个https配置只是针对luci，也就是80和443端口，其他服务包括docker提供的http服务不归luci管，https配置也是各服务自己负责的

closed 2023-01-10 by jjm2473

nevertoday, 2023-01-10:
没有很傻白甜的办法处理这种需求么？

lmlx66, 2023-01-10:
> 没有很傻白甜的办法处理这种需求么？

不处理，内网没必要用https

jjm2473, 2023-01-10:
可以参考这个做nginx反代，还有管理界面 https://blog.darkthread.net/blog/nginx-proxy-manager/

nevertoday, 2023-01-10:
> 可以参考这个做nginx反代，还有管理界面 https://blog.darkthread.net/blog/nginx-proxy-manager/

thank u so  much ~!!