istoreos/istoreos #536: softethervpn-server 在软件包安装了也无法使用，希望上软件中心

byteinfo, 2022-12-23:
softethervpn-server 在软件包安装了也无法使用，希望上软件中心

love5461, 2022-12-25:
上软件中心是不可能了，自己找第三方吧。  上完了容易整个固件系统又被屏蔽

love5461, 2022-12-25:
> softethervpn-server 在软件包安装了也无法使用，希望上软件中心

软件中心基本上不会再上线关于任何VPN的资料

byteinfo, 2022-12-26:
> > softethervpn-server 在软件包安装了也无法使用，希望上软件中心
> 
> 软件中心基本上不会再上线关于任何VPN的资料

这个不是科学上网的的VPN，这个软件无关科学上网，只是用来做虚拟局域网的，跟 OpenVPN 类似。

jjm2473, 2023-01-04:
softethervpn不会上线的，如果是openwrt官方仓库就有的东西，可以直接用opkg安装

zhangtaidong, 2023-01-11:
就等Softethervpn 上线 了。OpenVPN zerotier还是没有Softether好用。

zhangtaidong, 2023-01-11:
> softethervpn不会上线的，如果是openwrt官方仓库就有的东西，可以直接用opkg安装

关键就是Openwr官方库中的东西安装用不了才是个问题。用OP官方固件运行一点问题没有，唯独就是到了Istore用不了。

jjm2473, 2023-01-12:
@zhangtaidong 你是按这个文档配置的吗，出现什么问题？有没有日志？ https://openwrt.org/docs/guide-user/services/vpn/softethervpn/client

darkrain88, 2023-03-25:
r系统软件中心 可以安装， 用命令配置

zhangtaidong, 2023-03-25:
> @zhangtaidong 你是按这个文档配置的吗，出现什么问题？有没有日志？ https://openwrt.org/docs/guide-user/services/vpn/softethervpn/client   
现在已经搞定，可以正常安装运行。谢谢

80houco, 2023-04-28:
同求 命令安装 能说明下吗 不是中文 好费力啊 ，按上面 弄了一天。。还是不会

yixianghsy, 2023-11-05:
> Reference in n

> > @zhangtaidong 你是按这个文档配置的吗，出现什么问题？有没有日志？ https://openwrt.org/docs/guide-user/services/vpn/softethervpn/client
> > 现在已经搞定，可以正常安装运行。谢谢

按照那个文档 弄得吗？我也想弄好几天都弄不好