istoreos/istoreos #537: git: 'remote-https' is not a git command. See 'git --help'.

songleeCN, 2022-12-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：git: 'remote-https' is not a git command. See 'git --help'.

### (2) 路由器型号和固件版本

A：x86 iStoreOS 21.02.3 2022121613 / LuCI istoreos-21.02 branch git-22.332.46448-a322196

### (3) 详细日志和/或截图

A：root@SongLEE-Route:~# git clone https://github.com/shiunke/nvjdcdocker.git /root/nolanjdcr
Cloning into '/root/nolanjdcr'...
git: 'remote-https' is not a git command. See 'git --help'.

closed 2023-01-04 by jjm2473

RangerCui, 2023-08-07:
> 与istoreos无关，是git命令没安装完整

怎么解决这个问题呢？

jjm2473, 2023-08-07:
git的问题也要来这里讨论？看清楚，不是在istoreos里面执行git，是在电脑上执行git，电脑上执行git失败，居然来istoreos反馈？这是什么逻辑？git的问题不会谷歌搜索下吗？

RangerCui, 2023-08-07:
> git的问题也要来这里讨论？看清楚，不是在istoreos里面执行git，是在电脑上执行git，电脑上执行git失败，居然来istoreos反馈？这是什么逻辑？git的问题不会谷歌搜索下吗？

首先，搜git问题的是出现这个词条，然后就看到了你这个评论，既然你说了是什么原因，就简单问下，
其次，有尝试过重装git，和搜到的所有方式，并没有好的方式来解决git命令不完整问题，

不过，我觉得不算是讨论，只是你发了这个原因，觉得你可能知道如何解决，简单问下，知道或者不知道，都不要紧，不是很懂你为什么这么的火气？或者说其他的情绪波动？

而且就算是在非git的里面询问了这个问题，emmmm，算了，真是叨扰了！

jjm2473, 2023-08-11:
这里不是论坛OK？不要发无关的问题浪费开发者时间

jjm2473, 2023-08-11:
如果在openwrt里面需要支持git-https，参考这个  https://gist.github.com/mcandre/e5dfc4f1882d1925e8bee7b6586073d5

这不是谷歌下就能搜到的吗

zhanghaohao1013, 2023-09-13:
> 这里不是论坛OK？不要发无关的问题浪费开发者时间

> git的问题也要来这里讨论？看清楚，不是在istoreos里面执行git，是在电脑上执行git，电脑上执行git失败，居然来istoreos反馈？这是什么逻辑？git的问题不会谷歌搜索下吗？

纯路人，觉得你有点火大，建议降降火再说话。