istoreos/istoreos #550: 2022121613-x86-64版本旁路由模式网关配置不生效[可能存在的bug]

lmlx66, 2023-01-08:
### 环境
版本为 `istoreos-21.02.3-2022121613-x86-64-squashfs-combined.img.gz` ，使用VirtualBox桥接windows网卡进行测试学习。

### 问题描述

进入配置界面，家庭主路由ip192.168.123.1，windows桥接模式下依赖主路由dhcp获取192.168.123.123成功访问外网。更改静态ip为192.168.123.230，网关192.168.123.1（保证无误）无法ping通外网。

控制台输入 `route` 结果如下：

```
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
172.17.0.0      *               255.255.0.0     U     0      0        0 docker0
192.168.123.0   *               255.255.255.0   U     0      0        0 eth0
```

发现网关并未正确配置，输入以下指令配置网关

``` 
route add default gw 192.168.1.1
```

再次控制台输入 `route` 结果如下：

``` 
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.123.1   0.0.0.0         UG    0      0        0 eth0
172.17.0.0      *               255.255.0.0     U     0      0        0 docker0
192.168.123.0   *               255.255.255.0   U     0      0        0 eth0
```

之后可ping通外网，问题解决

### 目的

1. 提出bug：小白，并不清楚这个问题是否是潜在的bug。
2. 给出对应问题的解决方案

UntaggedRui, 2023-01-08:
我也遇到了这个问题，使用旁路由无法生效。

UntaggedRui, 2023-01-08:
[image]
我手动添加了三条规则也不行，请问该如何设置？

lmlx66, 2023-01-08:
> 

网络 -> 防火墙，最下面区域，wan口开启ip动态伪装。没有外网的问题使用我上述命令能够解决。

lmlx66, 2023-01-08:
很抱歉占用资源，原因是因为我直接跑的u盘镜像，并未重新挂载磁盘安装系统。

My-By, 2023-01-19:
主路由是*.123.1，然后网关填123.1，理应不是访问不了外网吗，还是我理解错误？

jjm2473, 2023-02-17:
配置旁路由以后，打开 “网络” - “接口”，编辑 “LAN”，切换到 “高级设置” 标签，看看有没有勾选 “使用默认网关”（应该要勾选）。同样，也看看 “WAN” 的 “使用默认网关”（应该是不勾选）

jjm2473, 2023-02-17:
还有，配置以后直接贴出 `/etc/config/network` 文件的内容。

jjm2473, 2023-03-16:
最新固件在开关旁路由模式时会确保只有一个接口勾选默认网关，应该就没问题了

closed 2023-03-16 by jjm2473