istoreos/istoreos #553: x86最新版磁盘分区一次后无法再分区

Devil523343732SKY, 2023-01-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

无法进行更多的分区，手动分区一次或者首页docker自动配置一次之后，无法再对硬盘分区挂载

A：
X86机器，固件istoreos-21.02.3-2022121613-x86-64-squashfs-combined.img.gz

A：

[image]

A：
[image]
[image]
[image]

jjm2473, 2023-01-10:
MBR分区表最多只有4个主分区，且目前这些分区工具不支持扩展分区。
如果需要4个分区以上，可以：
1. 使用efi固件
2. 手动使用命令创建扩展分区
3. 如果不是启动盘，那么改成GPT分区表即可

jjm2473, 2023-01-10:
https://zh.wikipedia.org/wiki/%E4%B8%BB%E5%BC%95%E5%AF%BC%E8%AE%B0%E5%BD%95#%E4%B8%BB%E5%BC%95%E5%AF%BC%E8%AE%B0%E5%BD%95%E4%B8%8E%E7%A1%AC%E7%9B%98%E5%88%86%E5%8C%BA

Devil523343732SKY, 2023-01-10:
> 

原来如此，谢谢大佬让我增长知识了

Devil523343732SKY, 2023-01-10:
@jjm2473 原来如此，谢谢大佬让我增长知识了，马上去改一下

closed 2023-01-16 by jjm2473