istoreos/istoreos #555: 突然进不去路由器主页了

huaxiac4, 2023-01-15:
今天突然就打不开了，打开主页就报错。但是本身路由功能都没有问题。可以正常上网。帮忙看下如何解决。很急

错误代码如下:
Status: 500 Internal Server Error /usr/lib/lua/luci/template.lua:97: Failed to execute template 'istorex/index'. A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'istorex/main'. A runtime error occurred: /usr/lib/lua/luci/template.lua:74: Failed to load template 'tasks/embed'. Error while parsing template '/usr/lib/lua/luci/view/tasks/embed.htm': No such file or directory stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:74: in function '__init__' /usr/lib/lua/luci/util.lua:65: in function 'Template' /usr/lib/lua/luci/dispatcher.lua:766: in function 'include' [string "/usr/lib/lua/luci/view/istorex/main.htm"]:1: in main chunk stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function (tail call): ? /usr/lib/lua/luci/controller/istorex.lua:74: in function 'istorex_template' [string "/usr/lib/lua/luci/view/istorex/index.htm"]:3: in main chunk stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function (tail call): ? /usr/lib/lua/luci/dispatcher.lua:1027: in function 'dispatch' /usr/lib/lua/luci/dispatcher.lua:478: in function
[image]

kevingraham03, 2023-01-26:
我也是同样的问题，功能正常，但是打不开管理界面
[image]

jjm2473, 2023-01-29:
可能安装过程中出现异常了，例如磁盘满了。使用终端或者ssh执行 `is-opkg install luci-lib-taskd` 看看

huaxiac4, 2023-01-30:
> 

没有等到您恢复了。我直接格盘重装了

closed 2023-01-30 by jjm2473

dunquzhiyi, 2023-02-03:
我也是这个问题
[image]

jjm2473, 2023-02-03:
@dunquzhiyi 明显不是同一个问题啊，看看自己有没有翻墙吧，或者访问的ip对不对

dunquzhiyi, 2023-02-03:
> @dunquzhiyi明显不是同一个问题啊，看自己有没有翻墙吧，或者访问者访问的ip对不对

ip是对的，我在设置端口转发的时候报错，刷新之后进不去了。设置端口转发的目的是想在公网访问管理页面，已经做了ddns。