istoreos/istoreos #557: 安装后拔去U盘重启报错i915 0000:00:02.0: Failed to load DMC firmware i915/kbl_dmc_ver1_04.bin. Disabling runtime power management.

MrNewOne, 2023-01-16:
### (1) 具体问题

A：安装后拔去U盘重启报错i915 0000:00:02.0: Failed to load DMC firmware i915/kbl_dmc_ver1_04.bin. Disabling runtime power management.

### (2) 路由器型号和固件版本

A：固件版本：istoreos-21.02.3-2022121613-x86-64-squashfs-combined-efi.img.gz
路由器：J4125 4口RTL8125B网卡
### (3) 详细日志和/或截图

[image]

>

L0tus0, 2023-01-16:
我在也遇到了同样的问题，据恩山一个帖子说是因为root device PARTUUID没有指向正确的引导分区所致。
我最后是安装1207的legacy版本解决的。

MrNewOne, 2023-01-16:
> 我也遇到了同样的问题，据恩山一个帖子说是因为root device PARTUUID没有指向正确的引导分区所致。 我最后是安装1207的legacy版本解决的。

多谢，那我试一下1207的efi版本

jjm2473, 2023-01-29:
如果是使用quickstart安装导致的问题，那之后会修复

jjm2473, 2023-02-05:
已修复

closed 2023-02-05 by jjm2473