istoreos/istoreos #561: r66s上的istoreos固件无法重启

5265325, 2023-01-22:
## 1.关于你要提交的问题
r66s上的istoreos固件无法重启

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：r66s烧录istoreos固件后，开机进入192.168.100.1，进入系统->重启页面直接重启，以太网断开连接后一直无法恢复，需要通过拔插电源来重启。尝试过r66s的出厂固件，并没有该现象。

### (2) 路由器型号和固件版本

A：istoreos-21.02.3-2022121613-r66s-squashfs.img.gz[https://fw.koolcenter.com/iStoreOS/r66s/istoreos-21.02.3-2022121613-r66s-squashfs.img.gz]

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2023-01-29:
目前已知部分比较新的sd卡会这样，如果有旧sd卡可以先试试。
麻烦提供下sd卡的品牌和型号，方便我们之后测试

rootphantomer, 2023-03-10:
> 目前已知部分比较新的sd卡会这样，如果有旧sd卡可以先试试。 麻烦提供下sd卡的品牌和型号，方便我们之后测试

最近刚买的  联想thinkplus 32g tf卡

jjm2473, 2023-03-10:
@rootphantomer  新固件已经修复了，等发布上线

closed 2023-03-10 by jjm2473