istoreos/istoreos #562: iStore自动安装网心云Docker

kingjoydom, 2023-01-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
iStore自动安装网心云docker

### (1) 具体问题

A：为啥我没有安装网心云会被自动安装，而且我删了这个容器或者停止这个容器都没用，会自动重新安装或者重新启动

### (2) 路由器型号和固件版本
[image]

A：

### (3) 详细日志和/或截图
[image]

A：

jjm2473, 2023-01-29:
看看服务菜单有没有网心云

kingjoydom, 2023-01-30:
没有

jjm2473, 2023-01-30:
ttnode是你安装的吗，看看安装脚本有没有把wxedge一起安装了，尤其看看定时任务和开机启动项

closed 2023-01-30 by jjm2473

piaoyz, 2024-09-01:
我在N4100小主机里字装istoreos22.03.7系统，在网心云界面，安装配置里点击安装时总是安装不上，已经接上240G的SSD盘并格式化成EXT4。请指导一下好吗？