istoreos/istoreos #565: Rockchip方案机型从高速sd卡启动后软重启失败

jjm2473, 2023-01-30:
目前已知r2s,r4s,r66s等机型使用较新型的高速sd卡启动以后，软重启会失败（硬重启没问题）。
相关问题在这个issue跟进，如果有测试固件也会发到这里。

如果你遇到相关问题，请跟帖提供问题机型和使用的sd卡的品牌型号。

相关issues：https://github.com/istoreos/istoreos/issues/526 https://github.com/istoreos/istoreos/issues/561 https://github.com/istoreos/istoreos/issues/543 https://github.com/istoreos/istoreos/issues/205

根据其他openwrt的补丁，可能是要给sd卡降速，重启才不会有问题。参考这些补丁：
https://github.com/coolsnowwolf/lede/blob/6fc2d00f257c988ccb27e618317a0c83f496f103/target/linux/rockchip/patches-5.4/107-nanopi-r4s-sd-signalling.patch
https://github.com/coolsnowwolf/lede/blob/6fc2d00f257c988ccb27e618317a0c83f496f103/target/linux/rockchip/patches-5.4/105-mmc-core-set-initial-signal-voltage-on-power-off.patch
https://github.com/openwrt/openwrt/blob/2b583ab8a74a6bd41f1c9629b3b66ba561d6fceb/target/linux/rockchip/patches-5.10/105-nanopi-r4s-sd-signalling.patch

jjm2473, 2023-01-30:
测试固件 https://fw.koolcenter.com/iStoreOS/alpha/

lc5900, 2023-02-18:
一样的问题，机型：R66s，SD卡：电犀牛官方的32G卡

jjm2473, 2023-02-19:
@lc5900 升级最新固件（20230217）看看还有没有这个问题

lc5900, 2023-02-19:
> @lc5900 升级最新固件（20230217）看看还有没有这个问题

还是不行，等了几分钟都连不上，还是得手动断电

jjm2473, 2023-03-09:
r66s测试固件 https://we.tl/t-2GQwjYxb5n

lc5900, 2023-03-09:
> r66s测试固件 https://we.tl/t-2GQwjYxb5n

这个可以了，辛苦大佬了

jjm2473, 2023-03-10:
r66s，r4s已修复。

r2s不确定是否有问题，也没看到其他固件的修复方案。

closed 2023-03-10 by jjm2473

QYKing233, 2023-03-15:
我看看

davidzhang12138, 2023-03-31:
r2c也遇到了 也是软重启不行 必须断电 每次升级后都需要断电重启

davidzhang12138, 2023-04-04:
整个包试试看？ 还是下次发包塞进去

jjm2473, 2023-04-05:
@davidzhang12138  试试这个 https://we.tl/t-rVTLz4J2sK

davidzhang12138, 2023-04-06:
> @davidzhang12138 试试这个 https://we.tl/t-rVTLz4J2sK

不行 还是无法软重启进入系统

jjm2473, 2023-04-06:
那跟r4s就不是同一个问题了

jjm2473, 2023-04-06:
@davidzhang12138 验证升级以后再软重启哦，不是升级时自动重启。

jjm2473, 2023-04-06:
或者就升级两次

davidzhang12138, 2023-04-06:
是的 升级完成后 断电重启了一次 然后软重启的 依然是不行

davidzhang12138, 2023-04-06:
我试下刷两遍看看

davidzhang12138, 2023-04-06:
不行 依然是无法软重启 整个系统重置后 也是一样

jjm2473, 2023-04-06:
OK，那r4s的方案没法用在r2s上，可能原因也不一样。

fanzhenhua0, 2025-08-12:
树莓派5和树莓派4也存在类似的问题，新装好的系统不进行任何设置，直接进行软重启，可以成功启动。
但是当对系统盘未分区的空间进行“分区并格式化”后，再通过软启动或者断电重启都会重启失败！
固件版本：istoreos-24.10.2-2025080813-raspberrypi-xxx
SD卡型号：SanDisk Extreme A2 128G；SanDisk Ultra A1 32G

jjm2473, 2025-08-20:
树莓派扩容无法启动的问题早就在更新日志说了，openwrt原版固件也一样的。你应该在扩容以后将硬盘id写到启动配置文件里。