istoreos/istoreos #566: r2s和r2c通用固件

jjm2473, 2023-01-30:
r2s和r2c使用的phy芯片不一致，并且mdio的addr不一样，按其他固件的方案是使用dtb来区分，但是需要各自生成固件。

根据这段代码
https://elixir.bootlin.com/linux/v5.4.230/source/drivers/of/of_mdio.c#L229
```
	/* Loop over the child nodes and register a phy_device for each phy */
	for_each_available_child_of_node(np, child) {
		addr = of_mdio_parse_addr(&mdio->dev, child);
		if (addr < 0) {
			scanphys = true;
			continue;
		}

		if (of_mdiobus_child_is_phy(child))
			rc = of_mdiobus_register_phy(mdio, child, addr);
		else
			rc = of_mdiobus_register_device(mdio, child, addr);

		if (rc == -ENODEV)
			dev_err(&mdio->dev,
				"MDIO device at address %d is missing.\n",
				addr);
		else if (rc)
			goto unregister;
	}

	if (!scanphys)
		return 0;

	/* auto scan for PHYs with empty reg property */
	for_each_available_child_of_node(np, child) {
		/* Skip PHYs with reg property set */
		if (of_find_property(child, "reg", NULL))
			continue;

		for (addr = 0; addr < PHY_MAX_ADDR; addr++) {
			/* skip already registered PHYs */
			if (mdiobus_is_registered_device(mdio, addr))
				continue;

			/* be noisy to encourage people to set reg property */
			dev_info(&mdio->dev, "scan phy %pOFn at address %i\n",
				 child, addr);

			if (of_mdiobus_child_is_phy(child)) {
				/* -ENODEV is the return code that PHYLIB has
				 * standardized on to indicate that bus
				 * scanning should continue.
				 */
				rc = of_mdiobus_register_phy(mdio, child, addr);
				if (!rc)
					break;
				if (rc != -ENODEV)
					goto unregister;
			}
		}
	}

	return 0;
```
将dtb节点的`compatible`改成 `compatible = "ethernet-phy-ieee802.3-c22";`，并且删掉`reg`字段，应该就可以让内核自动配置phy了。

相关文档：
https://www.kernel.org/doc/Documentation/devicetree/bindings/net/ethernet-phy.yaml

jjm2473, 2023-01-30:
测试固件： https://we.tl/t-Oa09YDrfRs

ks-bao, 2023-01-30:
提供另外一个测试固件的下载地址：
https://fw.koolcenter.com/iStoreOS/alpha/istoreos-r2c-test_nanopi-r2s-squashfs-sysupgrade.img.gz

jjm2473, 2023-01-31:
@bigdongdongCLUB  麻烦测试下新固件，同样是r2s和r2c： https://we.tl/t-N0NuDgYh6i

mynameisalice, 2023-01-31:
安装PW2显示依赖的软件包 iptables-mod-socket 在所有仓库都未提供，不清楚是不是缺少内核模块。

jjm2473, 2023-01-31:
@mynameisalice 这里只讨论r2s,r2c统一固件网络是否正常，其他问题搜索已有issue或者提新issue

picc745514, 2023-02-02:
R2C,网卡工作正常，500M宽带测试只能到400M。

jjm2473, 2023-02-02:
性能问题不是太离谱就行，可能是转发性能瓶颈。真要测试性能应该使用homebox，iperf等自建的服务测速，而不是依赖运营商带宽。

jjm2473, 2023-02-05:
融合固件已发布

closed 2023-02-05 by jjm2473

jjm2473, 2023-02-09:
> R2C,网卡工作正常，500M宽带测试只能到400M。

@picc745514 
试试这个固件 https://we.tl/t-mQKL9IoVKj ，看网速会不会快一些。还有启动以后执行 `dmesg | grep 'Set MAC tx_delay'` 看看输出什么

shoyuf, 2023-06-30:
> 融合固件已发布

@jjm2473 该固件是否只能用在r2c而无法烧录在r2c plus的emmc上使用？我目前烧录后sys常亮，wan\lan常暗看上去系统没有启动成功，是否烧录流程有什么特别的步骤？

jjm2473, 2023-07-01:
@shoyuf 没听说过带emmc的，只支持r2s/r2c，小众的型号除非openwrt自带机型，不然也不会适配的。

shoyuf, 2023-07-02:
> @shoyuf 没听说过带emmc的，只支持r2s/r2c，小众的型号除非openwrt自带机型，不然也不会适配的。

除了8gb的emmc，和 r2c 规格一样的，这个可以适配一下吗？http://wiki.friendlyelec.com/wiki/index.php/NanoPi_R2C_Plus/zh#NanoPi_R2C_Plus.E8.B5.84.E6.BA.90.E7.89.B9.E6.80.A7

jjm2473, 2023-07-02:
都说了小众的机型不会移植，代码是开源的，可以自己修改编译。或者等openwrt支持了这个机型，我们可能也会支持

harveyhonorg, 2023-10-21:
设备型号  FriendlyElec NanoPi R5S
固件版本  iStoreOS 22.03.5 2023102014
内核版本  5.10.176

最近这个docker蹦的 大家有遇到么？
https://github.com/istoreos/istoreos/assets/50670499/a87022ab-3f0e-4e0c-9e8d-a719f27df3a0