istoreos/istoreos #569: 再提一下，X86版的固件，请求增加主机的实时流量

sooway, 2023-01-31:
有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：再提一下，X86版的固件，首页真没有实时流量，软件中心也没有，翻遍菜单也没有，这个问题有人提了几次了，麻烦核实下，谢谢开发者，要显示局域网内，各个主机的实时流量。

### (2) 路由器型号和固件版本

A：X86版，从始至终的版本都没有

### (3) 详细日志和/或截图

要下面这种样子的
[image]

jjm2473, 2023-01-31:
这个插件叫什么

jjm2473, 2023-01-31:
istore里面不是有个“主机流量监控”插件吗？

sooway, 2023-02-03:
“主机流量监控”不好用，上面那个插件名字好像是wrtbwmon

Bender1337, 2023-02-06:
如果能加上这个功能就太好了。
“主机流量监控”这个插件根本不好用

My-By, 2023-02-14:
主机流量监控确实是累赘，真想看看哪个设备跑流量，都看不好。

jjm2473, 2023-03-02:
istore上架了 `实时流量` ，这个插件的更新频率是5秒一次，足够实时监控了。

`主机流量监控` 只是数据更新频率慢一点，最快30秒一次，并不是不能用，而且统计数据更丰富。

closed 2023-03-02 by jjm2473

sooway, 2023-03-08:
> istore上架了 `实时流量` ，这个插件的更新频率是5秒一次，足够实时监控了。
> 
> `主机流量监控` 只是数据更新频率慢一点，最快30秒一次，并不是不能用，而且统计数据更丰富。

谢谢！！！！！