istoreos/istoreos #570: R2C测试固件内核iptables-mod-socket不可用

mynameisalice, 2023-01-31:
### (1) 具体问题

A：使用R2C测试固件，网络正常，安装软件包显示  `iptables-mod-socket` 不可用

### (2) 路由器型号和固件版本

A：R2C `istoreos-r2c-test-20230131_nanopi-r2s-squashfs-sysupgrade.img.gz`

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2023-02-03:
应该是官方仓库本来就没这个包吧

mynameisalice, 2023-02-03:
> 应该是官方仓库本来就没这个包吧

嗯，我使用第三方软件源也安装不上，可能是编译的时候没加进去。

jjm2473, 2023-02-03:
既然21版本的官方仓库都没有这个包，那就等以后升级在看有没有吧，先用passwall

jjm2473, 2023-02-04:
@mynameisalice  安装这两个空包

* https://istore.linkease.com/repo/all/meta/kmod-ipt-socket_0.00_all.ipk
* https://istore.linkease.com/repo/all/meta/iptables-mod-socket_0.00_all.ipk

然后应该就能安装passwall2了

closed 2023-02-05 by mynameisalice

jjm2473, 2023-03-03:
升级istore以后就可以安装了