istoreos/istoreos #572: 增加功能需求

Alstonsz, 2023-02-02:
能否在软件中心增加 HTTPS设置，增加自动申请SSL证书功能；方便外网使用HTTPS，访问。

jjm2473, 2023-02-03:
https证书没那么简单就能申请，必须要有自己的域名，能自定义dns解析，如果能满足条件，用acme就可以申请证书。

Alstonsz, 2023-02-07:
如果在用户有自己域名、做了DNS解析都满足的情况下，是否可以像爱快那样：如下图
[image: image.png]
[image: image.png]
这样方便很多。
通过SSH的方式，挺麻烦还不容易成功；折腾了好久都没搞通。
证书通过acme申请下来了，但一直Https访问不了（做了端口转发，添加了Uhttpd 443端口）；
实现这个功能主要是在外网摄像头、Plex(http模式下串流转码问题）
希望能添加这样的一件工具，方便使用。

jjm2473, 2023-02-07:
看不到图片。既然证书都申请下来了，不是可以直接用吗？不能访问可能是运营商拦截，换成10000以上的端口试试

gedoor, 2023-07-23:
阿里云有免费的证书,就是一次只能一年,能自动重新申请就好了

CG1995, 2024-03-14:
有域名，https端口改成常见的443和80以外的端口，或者直接lucky、nginx反代出来就好了，这已经非常简单了，可以完全纯界面操作，再说acme也有luci-app，界面操作也很简单，如果istoreos直接整合未免太臃肿，这样的话什么细枝末节的东西都要开发未免类似开发者了。

jjm2473, 2024-03-15:
推荐用 linuxserver/swag 容器，集成了自动证书申请/更新，nginx反代，banip等功能

pili-book, 2024-03-18:
> https证书没那么简单就能申请，必须要有自己的域名，能自定义dns解析，如果能满足条件，用acme就可以申请证书。

有没有r6s适用的acme 插件，istore商店里没这个插件，要是能集成到自带的商店就更好了，网上看到有但是下载下来不能用。。。。不知道为什么

evoking6, 2024-07-03:
你可能在找nginx manager