istoreos/istoreos #578: 请求添加对NanoPi R5C的支持

tcbt999, 2023-02-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：NanoPi R5C 刷入R5S版本无法启动。

### (2) 路由器型号和固件版本

A：NanoPi R5C   openwrt  R5S所有版本

### (3) 详细日志和/或截图

A：可以刷入，但无法启动。

jjm2473, 2023-02-07:
官方的R5C和R5S使用同一个固件吗

tcbt999, 2023-02-09:
有的可以编译成一个固件，比如官方的friendlywrt，骷髅头是分下来的。

GreenTeodoro839, 2023-02-11:
> 官方的R5C和R5S使用同一个固件吗

跟r2s r2c差不多，网口配置不一样

tcbt999, 2023-02-12:
是的

tcbt999, 2023-02-12:
官方的R5C和R5S确实使用同一个固件

jjm2473, 2023-02-13:
如果只是PHY芯片不一样，那就简单了

jjm2473, 2023-02-13:
如果只是PHY芯片不一样，应该不至于不能启动

GreenTeodoro839, 2023-02-13:
> 如果只是PHY芯片不一样，应该不至于不能启动

我也不太明白，可以参考一下骷颅头的r5c固件

antbuster, 2023-02-17:
R5S固件可以用，但是一个USB口无法识别，LED灯无用，LAN口错乱。还有相比官方固件，发热严重

jjm2473, 2023-02-18:
> R5S固件可以用，但是一个USB口无法识别，LED灯无用，LAN口错乱。还有相比官方固件，发热严重

发热是因为没驱动风扇？

bearcolin, 2023-02-20:
5C比5S还多了个m.2的无线网卡 rtl8822ce

v1006945671, 2023-03-08:
我刷了，可以启动，但是灯错乱的，多了一个不存在的网口

jjm2473, 2023-06-09:
最新固件已经支持了，跟R5S同一个固件，如果之前刷了R5S的固件，要全新刷机才有效。

closed 2023-06-09 by jjm2473