istoreos/istoreos #579: 希望增加对UVC 相机设备的支持

tomssy, 2023-02-06:
希望启用内核中的kmod-usb-video，kmod-video-uvc等支持，这一功能适合使用USB类视频输入设备如相机等，跑MJPGstreamer当简易监控/KVM等
目前插入UVC设备无法生成/dev/vedio*类似节点

jjm2473, 2023-02-07:
哪个机型固件？如果不是3568方案的，都可以直接从软件包或者opkg安装内核模块

tomssy, 2023-02-07:
> 哪个机型固件？如果不是3568方案的，都可以直接从软件包或者opkg安装内核模块

标准x86 64位固件。
已使用opkg install kmod-video-core kmod-video-uvc kmod-i2c-core libv4l 安装相关依赖，均提示up to date，但是插入uvc设备后，并不能正常生成/dev/vedio*型设备节点，查阅日志无uvc相关信息。
进一步测式发现，手动尝试加载insmod /lib/modules/5.4.188/uvcvideo.ko提示failed to insert。有资料指出uvcvideo模块前置依赖videobuf2-vmalloc，尝试insmod /lib/modules/5.4.188/videobuf2-vmalloc.ko同样报 failed to insert。
故请求支持，是否是因为内核中尚有其他依赖未被选中启用？

jjm2473, 2023-02-08:
不知道，insmod失败以后看看内核日志，应该会告诉你为什么失败

tomssy, 2023-02-08:
> 不知道，insmod失败以后看看内核日志，应该会告诉你为什么失败

insmod uvcviedeo.ko 后日志为 uvcvideo: Unknown symbol vb2_vmalloc_memops (err -2)

insmod videobuf2-vmalloc.ko后日志为 videobuf2_vmalloc: Unknown symbol vb2_common_vm_ops (err -2)
videobuf2_vmalloc: Unknown symbol frame_vector_to_pfns (err -2)
videobuf2_vmalloc: Unknown symbol vb2_create_framevec (err -2)
videobuf2_vmalloc: Unknown symbol vb2_destroy_framevec (err -2)
videobuf2_vmalloc: Unknown symbol frame_vector_to_pages (err -2)

另外，执行modprobe /lib/modules/5.4.188/uvcvideo.ko提示：
3 modules could not be probed
- uvcvideo
- videobuf2-memops
- videobuf2-vmalloc

再补充 insmod videobuf2-memops.ko后日志为 videobuf2_memops: Unknown symbol put_vaddr_frames (err -2)
videobuf2_memops: Unknown symbol get_vaddr_frames (err -2)
videobuf2_memops: Unknown symbol frame_vector_destroy (err -2)
videobuf2_memops: Unknown symbol frame_vector_create (err -2)
videobuf2_memops: Unknown symbol put_vaddr_frames (err -2)
videobuf2_memops: Unknown symbol get_vaddr_frames (err -2)
videobuf2_memops: Unknown symbol frame_vector_destroy (err -2)
videobuf2_memops: Unknown symbol frame_vector_create (err -2)

水平有限，只能怀疑内核的v4l2相关部分尚有问题

jjm2473, 2023-02-10:
可能是内核没开 frame_vector 特性，所以 videobuf2-memops.ko 加载失败，之后的依赖就更不用说了

tomssy, 2023-02-10:
> 可能是内核没开 frame_vector 特性，所以 videobuf2-memops.ko 加载失败，之后的依赖就更不用说了

嗯，只能看开发这边，有没有后续计划添加支持了

jjm2473, 2023-02-17:
下个版本的固件会支持，预计这两天发布

closed 2023-02-17 by jjm2473