istoreos/istoreos #582: iptv 单线复用 vlan出错

tx5866, 2023-02-09:
## 2. 详细叙述

### (1) 具体问题

A：iptv 单线复用 vlan出错

### (2) 路由器型号和固件版本

A：路由型号 N5105

### (3) 详细日志和/或截图

A：路由配置：
[image]
配置2:
[image]
配置3：
[image]

光猫配置
[image]
[image]
[image]

network的配置：
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd8c:3cf9:9a05::/48'

config interface 'wan'
        option device 'eth0'
        option proto 'pppoe'
        option username '075504086227@163.gd'
        option password '81223939'

config interface 'wan6'
        option device 'eth0'
        option proto 'dhcpv6'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1'
        list ports 'eth2'
        list ports 'eth3'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option defaultroute '0'
        list ipaddr '192.168.2.1/24'

config interface 'docker'
        option device 'docker0'
        option proto 'none'
        option auto '0'

config device
        option type '8021q'
        option ifname 'eth0'
        option vid '2'
        option name 'eth0.2'

config device
        option type '8021q'
        option ifname 'eth0'
        option vid '3'
        option name 'eth0.3'

root@iStoreOS:~# vim  /etc/config/network 
root@iStoreOS:~# vim  /etc/config/network 
root@iStoreOS:~# /etc/init.d/network  reload
root@iStoreOS:~# vim  /etc/config/network 
root@iStoreOS:~# vim  /etc/config/network 
root@iStoreOS:~# vim  /etc/config/network 
root@iStoreOS:~# /etc/init.d/network  reload
root@iStoreOS:~# /etc/init.d/network  reload
root@iStoreOS:~# cat /etc/config/network 

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd8c:3cf9:9a05::/48'

config interface 'wan'
        option proto 'pppoe'
        option username '075504086227@163.gd'
        option password '81223939'
        option device 'eth0'

config interface 'wan6'
        option device 'eth0'
        option proto 'dhcpv6'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1'
        list ports 'eth2'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option defaultroute '0'
        list ipaddr '192.168.2.1/24'

config interface 'docker'
        option device 'docker0'
        option proto 'none'
        option auto '0'

config device
        option type '8021q'
        option ifname 'eth0'
        option vid '2'
        option name 'eth0.2'

config device
        option type '8021q'
        option ifname 'eth0'
        option vid '3'
        option name 'eth0.3'

config interface 'lan1'
        option proto 'static'
        option device 'eth3'

config device
        option type 'bridge'
        option name 'iptv'
        list ports 'eth0.3'
        list ports 'eth3'

config interface 'iptv'
        option proto 'none'
        option device **'iptv'**

jjm2473, 2023-02-17:
eth3已经作为一个逻辑接口了？那就不能作为桥接了。

jjm2473, 2023-02-17:
也就是删掉
```
config interface 'lan1'
    option proto 'static'
    option device 'eth3'
```

closed 2023-02-18 by jjm2473

locis08, 2023-05-06:
请问问题解决了吗？我也遇到这个问题，不知道怎么解决。感谢