istoreos/istoreos #583: istore商店安装alist后无法启动

ray1888, 2023-02-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
alist安装后无法启动，并且没有日志输出

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
安装好istoreOS后，在软件商店安装alist。alist修改配置为启动后，一直没启动成功，并且没有日志输出

### (1) 具体问题

A：alist进程无法启动

### (2) 路由器型号和固件版本

A：AZW Gemini J45

### (3) 详细日志和/或截图

A：
[image]
[image]

ray1888, 2023-02-10:
后续：使用Alist server 命令是可以启动成功的，辛苦看一下启动和守护脚本的问题

GD2021, 2023-02-11:
我的树莓派4B也是无法启动，没有任何报错

GD2021, 2023-02-11:
输入nohup alist server &  这个命令后，成功启动

ray1888, 2023-02-12:
辛苦官方的同学看一下alist的service脚本哪里出问题了，因为不可能每次重启后都要靠nohup来继续启动的

ray1888, 2023-02-12:
直接使用命令 /bin/sh -x  来执行/etc/init.d/alist 中的 start_service 函数，执行结果如下
[image]

ray1888, 2023-02-14:
@jjm2473 辛苦有空看问题？不确定是alist的bug还是软件商店提供版本的问题（比心）

jjm2473, 2023-02-14:
ipk应该就是从官方复制的，等alist修复阿里云盘api以后再一起更新吧

ray1888, 2023-02-14:
@jjm2473 alist 在github发布的release是没有ipk的，是istoreos这边通过源码自己打包的吗？

ray1888, 2023-02-14:
@jjm2473 还是你们仓库的版本是从https://github.com/sbwml/luci-app-alist/releases/tag/v3.10.1 这里拉的？

jjm2473, 2023-02-16:
> @jjm2473 还是你们仓库的版本是从https://github.com/sbwml/luci-app-alist/releases/tag/v3.10.1 这里拉的？

对，直接从这里下载的ipk

ray1888, 2023-02-17:
@jjm2473 我这边尝试了一下，把从商店的卸载之后，然后直接手动安装ipk，nohup 可以拉起来，而且能稳定运行。用商店的版本加守护脚本使用nohup，会无端自动退出，并且不会返回任何日志。感觉还是alistd service的 start或者 restart那里有问题。但是我不太熟openwrt怎么调试这个。普通linux上面直接sh -x 就可以调试了，这个可能还是要你这边看一下

jjm2473, 2023-02-17:
@ray1888 我测试了可以启动，你用的是什么固件？

jjm2473, 2023-02-17:
https://github.com/sbwml/luci-app-alist/blob/master/luci-app-alist/root/etc/init.d/alist

alist的启动脚本是使用procd守护进程的，不会因为hup信号而退出。你还是直接用`/etc/init.d/alist start`看看能不能启动吧

jjm2473, 2023-02-17:
@ray1888 你是不是不了解procd的机制，为什么会用nohup？nohup用在什么地方，把用到nohup的命令发出来看看

hury88, 2023-02-17:
我也遇到通道同样的问题直接安装alist打开后无响应 命令行中 alist server启动后就能访问了，用的最新版istoreos-21.02.3-2023020316-x86-64-squashfs-combined.img.gz

jjm2473, 2023-02-18:
@hury88 你的也是显示“未运行”？

babylife, 2023-02-20:
一样的问题，我也碰到了，使用alist start可以正常启动访问管理页面。
纯干净的itstore OS 配置向导配置旁路由模式就可以重现问题，已经重置多次。 
环境单网卡旁路由。错误日志如下。
Mon Feb 20 08:40:42 2023 daemon.err alist[16389]: [36mINFO[0m[2023-02-20 00:40:42] start server @ 192.168.50.100/24:5244
Mon Feb 20 08:40:42 2023 daemon.err alist[16389]: [31mFATA[0m[2023-02-20 00:40:42] failed to start: listen tcp: lookup 192.168.50.100/24: no such host
Mon Feb 20 08:40:47 2023 daemon.err alist[16434]: [36mINFO[0m[2023-02-20 00:40:47] reading config file: /etc/alist/config.json
Mon Feb 20 08:40:47 2023 daemon.err alist[16434]: [36mINFO[0m[2023-02-20 00:40:47] load config from env with prefix: ALIST_
Mon Feb 20 08:40:47 2023 daemon.err alist[16434]: [36mINFO[0m[2023-02-20 00:40:47] init logrus...
Mon Feb 20 08:40:47 2023 daemon.err alist[16434]: [36mINFO[0m[2023-02-20 00:40:47] start server @ 192.168.50.100/24:5244
Mon Feb 20 08:40:47 2023 daemon.err alist[16434]: [31mFATA[0m[2023-02-20 00:40:47] failed to start: listen tcp: lookup 192.168.50.100/24: no such host
Mon Feb 20 08:40:52 2023 daemon.err alist[16456]: [36mINFO[0m[2023-02-20 00:40:52] reading config file: /etc/alist/config.json
Mon Feb 20 08:40:52 2023 daemon.err alist[16456]: [36mINFO[0m[2023-02-20 00:40:52] load config from env with prefix: ALIST_
Mon Feb 20 08:40:52 2023 daemon.err alist[16456]: [36mINFO[0m[2023-02-20 00:40:52] init logrus...
Mon Feb 20 08:40:52 2023 daemon.err alist[16456]: [36mINFO[0m[2023-02-20 00:40:52] start server @ 192.168.50.100/24:5244
Mon Feb 20 08:40:52 2023 daemon.err alist[16456]: [31mFATA[0m[2023-02-20 00:40:52] failed to start: listen tcp: lookup 192.168.50.100/24: no such host
Mon Feb 20 08:40:57 2023 daemon.err alist[16478]: [36mINFO[0m[2023-02-20 00:40:57] reading config file: /etc/alist/config.json
Mon Feb 20 08:40:57 2023 daemon.err alist[16478]: [36mINFO[0m[2023-02-20 00:40:57] load config from env with prefix: ALIST_
Mon Feb 20 08:40:57 2023 daemon.err alist[16478]: [36mINFO[0m[2023-02-20 00:40:57] init logrus...
Mon Feb 20 08:40:57 2023 daemon.err alist[16478]: [36mINFO[0m[2023-02-20 00:40:57] start server @ 192.168.50.100/24:5244
Mon Feb 20 08:40:57 2023 daemon.err alist[16478]: [31mFATA[0m[2023-02-20 00:40:57] failed to start: listen tcp: lookup 192.168.50.100/24: no such host
Mon Feb 20 08:41:02 2023 daemon.err alist[16499]: [36mINFO[0m[2023-02-20 00:41:02] reading config file: /etc/alist/config.json
Mon Feb 20 08:41:02 2023 daemon.err alist[16499]: [36mINFO[0m[2023-02-20 00:41:02] load config from env with prefix: ALIST_
Mon Feb 20 08:41:02 2023 daemon.err alist[16499]: [36mINFO[0m[2023-02-20 00:41:02] init logrus...
Mon Feb 20 08:41:02 2023 daemon.err alist[16499]: [36mINFO[0m[2023-02-20 00:41:02] start server @ 192.168.50.100/24:5244
Mon Feb 20 08:41:02 2023 daemon.err alist[16499]: [36mINFO[0m[2023-02-20 00:41:02] Aria2 not ready.
Mon Feb 20 08:41:02 2023 daemon.err alist[16499]: [36mINFO[0m[2023-02-20 00:41:02] qbittorrent not ready.
Mon Feb 20 08:41:02 2023 daemon.err alist[16499]: [31mFATA[0m[2023-02-20 00:41:02] failed to start: listen tcp: lookup 192.168.50.100/24: no such host
Mon Feb 20 08:41:02 2023 daemon.info procd: Instance alist::alist s in a crash loop 6 crashes, 0 seconds since last crash

使用命令成功执行：
[image]

ray1888, 2023-02-20:
> @ray1888 我测试了可以启动，你用的是什么固件？

[image]

ray1888, 2023-02-20:
> @ray1888 你是不是不了解procd的机制，为什么会用nohup？nohup用在什么地方，把用到nohup的命令发出来看看

不太熟悉procd的机制，就直接是`nohup alist server &`

jjm2473, 2023-02-20:
@babylife 根据你发的日志，你遇到的问题我已经给上游提pr了 https://github.com/sbwml/luci-app-alist/pull/49 ，等上游发布ipk以后就可以了

jjm2473, 2023-02-20:
@ray1888 要不你也看看系统日志？搜索“failed to start: listen tcp:”看是不是有同样的错误

jjm2473, 2023-02-20:
临时的解决方案是勾选“允许从外网访问”

jjm2473, 2023-02-22:
alist更新了，应该没问题了

closed 2023-02-22 by jjm2473

hubuguiqu, 2023-06-24:
新三路由器里安装了最新版本，同样出现上述问题，使用SSH客户端运行nohup alist server &后显示已运行，但是打不开web页面，并且退出SSH客户端后马上变为Alist未运行。求解！

jjm2473, 2023-07-17:
搞笑吗，还新三，新三是mips架构的吧，我们可没支持mips，如果是通过第三方仓库安装的，那就去问第三方仓库的维护者

hawkli, 2024-06-06:
看了一下，这个ISSUE竟然是一年前的了，我的ESXI的VM里装的istoreos，alist版本是istore商店里最新的3.33.0，然而还是有这样的问题。
我是单网卡旁由模式，试过打开外网访问这个，没用。只能命令行开启

jjm2473, 2024-06-06:
@hawkli 反馈给 https://github.com/sbwml/luci-app-alist/issue

j3628379, 2024-06-08:
商店安装的alist，之前一直正常，今天突然就打不开了，用 alist server命令可以打开，但是无法登录，提示密码错误，重置密码也不行
[image]

zhouLife, 2024-06-12:
输入nohup alist server & 这个命令后，成功启动。**关闭终端 直接挂了** 。。。。

jjm2473, 2024-06-17:
恢复出厂设置，然后安装alist，看看有没有问题

scrymile, 2024-06-19:
> 商店安装的alist，之前一直正常，今天突然就打不开了，用 alist server命令可以打开，但是无法登录，提示密码错误，重置密码也不行

跟你一样的问题，这个开发者都会用他没问题来回答。反馈是没有用的

hawkli, 2024-06-19:
> > 商店安装的alist，之前一直正常，今天突然就打不开了，用 alist server命令可以打开，但是无法登录，提示密码错误，重置密码也不行
> 
> 跟你一样的问题，这个开发者都会用他没问题来回答。反馈是没有用的

这个问题今天我忽然好象解决了，还没完全确认，但感觉是解决的。
大致的情况是我重启了IstoreOS，alist启动了，但是后台是登录不上去，仔细分析发现系统里似乎装了2个alist，或者至少是配置有二套的，导致系统自动启动的配置文件admin密码我一直不知道，也就是启动也进不去，而用alist server启动后，改密码出来的是另一个alist。（系统只能正常运行一个alist，不然5244端口会争用起不来的）
我想了一下，在界面上重置了密码，并把最上面的启动那个勾去掉，保存。再勾上，保存。等于在界面上把系统自动启动的alist强制改了密码并重启，然后登录上去了，进去以后所有配置都是空的，之前我用命令行跑起来的alist的配置全没了，我再导入本地配置，再改密码和使用，就全正常了。

j3628379, 2024-06-24:
我也解决了，办法是直接把istoreOS恢复出厂设置，再用docker重装alist，这下就没问题了.......
暂时不清楚到底之前是什么原因导致的，总之现在能自己用docker装的我都自己装，不用istore里的。

zhouLife, 2024-06-24:
docker装的是host模式吗？ 我用bridge模式能正常安装但是添加网盘会提示错误timeout 
host模式就正常

j3628379, 2024-06-25:
bridge模式，我只有微力同步用的是host模式，bridge大多数时候没法和局域网内的电脑直连，就改成host模式了。alist用bridge模式没有问题

gaoshulaoshi, 2024-08-15:
我卸载掉从istore os商店一键安装的版本从这个（https://dl.openwrt.ai）仓库重新安装后，设备重启alist也不会挂了