istoreos/istoreos #584: uu加速器手机app提示组件初始化失败

cero-ss, 2023-02-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
uu加速器启动失败
### (1) 具体问题

A：安装商店的uu加速器后，显示的是运行中，手机app也能正常识别，但是就是加速失败

### (2) 路由器型号和固件版本

A：NanoPi R5S  iStoreOS 21.02.3 2023020315

### (3) 详细日志和/或截图

A：
[image]
[image]

cero-ss, 2023-02-10:
我用商店和官网提供的openwrt插件安装方法都试了，结果是一样的加速失败

jjm2473, 2023-02-17:
lan口是桥接的吗？
还有官方的openwrt安装方式有任何报错吗？

cero-ss, 2023-02-18:
> lan口是桥接的吗？ 还有官方的openwrt安装方式有任何报错吗？

走旁路由的，线是插在lan口，官方的安装方法没有报错，手机都能识别到，但是加速时就会提示错误

jjm2473, 2023-02-19:
可能是uu不支持旁路由模式？旁路由情况下，lan口和网关都是同一个接口，可能uu不支持。

jjm2473, 2023-02-19:
如果uu不支持旁路由，那应该给他们反馈bug

cero-ss, 2023-02-19:
> 可能是uu不支持旁路由模式？旁路由情况下，lan口和网关都是同一个接口，可能uu不支持。

我用过其他固件，其它的固件uu是能正常使用的，和是哪个口没关系，也不是旁路由的问题

jjm2473, 2023-02-19:
我看了下插件的代码，确实都是瞎写的，固件插件作者不懂shell脚本，更不懂openwrt。

jjm2473, 2023-02-22:
更新uu插件看看还有没有问题

cero-ss, 2023-02-22:
> 更新uu插件看看还有没有问题

我刚刚从官网下载了最新的固件重刷，UU依旧加速失败，和uu的版本应该没什么关系

jjm2473, 2023-02-23:
@cero-ss 运行以后执行在istoreos里面执行 `netstat -anp|grep uuplugin` 看看

jjm2473, 2023-02-23:
可能是uu插件不认识新的openwrt网络配置文件

jjm2473, 2023-02-23:
我用r68s安装绑定，没有报错。uu插件没有日志，所以没法排查，你还是给uu客服反馈吧，他们肯定能提取日志的

jinxiao, 2023-03-03:
给网友弄了一下，发现类似问题。
看看启动日志的时候有没有遇到kernel module nfnetlink cannot be found之类的字眼。

jjm2473, 2023-03-03:
问下uu的客服，是不是只依赖tun驱动。

还有，是不是只在istoreos，且旁路由模式才有问题。如果是，那么应该是因为uu不支持新的openwrt网络配置文件。你们说的其他固件，只要不是官方openwrt，基本上还是使用luci 17版本，openwrt网络配置文件也是老的格式，自然没问题。

jjm2473, 2023-03-03:
我已经给uu客服反馈过了，但是他们没联系我，而且我也不是uu的用户，没法帮他们验证

jjm2473, 2023-03-05:
@cero-ss istoreos默认也是用旧的格式啊，只是在首页配置旁路由或者改lan口IP以后会使用新的格式，在“网络”-“接口”里面也可以改成新的格式，只要没执行这些操作，就还是旧格式

jjm2473, 2023-03-05:
执行 `cat /etc/config/network` 看看

jjm2473, 2023-03-05:
还有，可以试试恢复出厂设置以后，不要用首页的网络向导和局域网ip设置，而是到“网络”-“接口”那边手动配置，然后看看uu能不能用

cero-ss, 2023-03-05:
> 还有，可以试试恢复出厂设置以后，不要用首页的网络向导和局域网ip设置，而是到“网络”-“接口”那边手动配置，然后看看uu能不能用

我又重新刷了一遍固件，然后照你说的手动配置，没用首页的，依旧不行，执行代码也没用，重启后再执行你给的代码也不行

NewbiePanda, 2023-03-13:
我也碰到了一摸一样的问题。。。
插件能够正常运行，手机上 一旦链接到加速节点后就 报错 路由器网络组件初始化失败。
奇怪的是，我在搬家前 还能用的好好的，搬家后就出现这个问题了。。醉了

NewbiePanda, 2023-03-13:
> 我也碰到了一摸一样的问题。。。 插件能够正常运行，手机上 一旦链接到加速节点后就 报错 路由器网络组件初始化失败。 奇怪的是，我在搬家前 还能用的好好的，搬家后就出现这个问题了。。醉了

openwrt的配置都是一摸一样的，宽带迁移 配置没变过

jjm2473, 2023-03-13:
反馈给uu了，等他们排查吧

NewbiePanda, 2023-03-16:
解决了，应该是内核组件kmod-tun没有成功安装。验证是否成功安装可以通过ls /dev/net/tun查看。
如果没有这个文件的话，可以.
1.opkg remove kmod-tun
2.opkg update
3.opkg install kmod-tun重新安装就可以

jjm2473, 2023-03-16:
@NewbiePanda istoreos默认就集成了这个，可能你用的就不是istoreos，所以你的问题跟其他人的不一样。

Tanamaz, 2023-03-20:
我也遇到了一样的问题，
报错 路由器网络组件初始化失败。
反馈给UU客服了。

NewbiePanda, 2023-03-20:
😂我就是用的istoreos。默认集成了也不能代表启动失败啊，我回答的这个方法很有参考性，请不要淹没了。。

| |
***@***.***
|
|
***@***.***
|

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2023年03月16日 18:04 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [istoreos/istoreos] uu加速器手机app提示组件初始化失败 (Issue #584) |

@NewbiePanda istoreos默认就集成了这个，可能你用的就不是istoreos，所以你的问题跟其他人的不一样。

Tanamaz, 2023-03-20:
> 

我看了有安装kmod-tun

```

BusyBox v1.33.2 (2023-03-16 07:33:42 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 iStoreOS 21.02.3, 2023032001
 -----------------------------------------------------
root@iStoreOS:~# opkg list-installed | grep kmod-tun
kmod-tun - 4.19.238-1
root@iStoreOS:~# 
```
还是提示 系统网络组件初始化失败
这是系统日志
```
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Joining mDNS multicast group on interface tun163.IPv4 with address 172.19.163.1.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: New relevant interface tun163.IPv4 for mDNS.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Registering new address record for 172.19.163.1 on tun163.IPv4.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Joining mDNS multicast group on interface tun163.IPv6 with address fe80::adef:b896:4370:c9b1.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: New relevant interface tun163.IPv6 for mDNS.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Registering new address record for fe80::adef:b896:4370:c9b1 on tun163.*.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Withdrawing address record for 172.19.163.1 on tun163.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Leaving mDNS multicast group on interface tun163.IPv4 with address 172.19.163.1.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Interface tun163.IPv4 no longer relevant for mDNS.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Joining mDNS multicast group on interface tun163.IPv4 with address 172.19.163.1.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: New relevant interface tun163.IPv4 for mDNS.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Registering new address record for 172.19.163.1 on tun163.IPv4.
Mon Mar 20 14:22:19 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:19 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:19 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:19 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Interface tun163.IPv6 no longer relevant for mDNS.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Leaving mDNS multicast group on interface tun163.IPv6 with address fe80::adef:b896:4370:c9b1.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Interface tun163.IPv4 no longer relevant for mDNS.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Leaving mDNS multicast group on interface tun163.IPv4 with address 172.19.163.1.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Withdrawing address record for fe80::adef:b896:4370:c9b1 on tun163.
Mon Mar 20 14:22:19 2023 daemon.info avahi-daemon[3521]: Withdrawing address record for 172.19.163.1 on tun163.
Mon Mar 20 14:22:19 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:19 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:19 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:19 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:24 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:24 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:24 2023 daemon.info avahi-daemon[3521]: Joining mDNS multicast group on interface tun163.IPv4 with address 172.19.163.1.
Mon Mar 20 14:22:24 2023 daemon.info avahi-daemon[3521]: New relevant interface tun163.IPv4 for mDNS.
Mon Mar 20 14:22:24 2023 daemon.info avahi-daemon[3521]: Registering new address record for 172.19.163.1 on tun163.IPv4.
Mon Mar 20 14:22:24 2023 daemon.info avahi-daemon[3521]: Joining mDNS multicast group on interface tun163.IPv6 with address fe80::9944:a79c:3812:1e1b.
Mon Mar 20 14:22:24 2023 daemon.info avahi-daemon[3521]: New relevant interface tun163.IPv6 for mDNS.
Mon Mar 20 14:22:24 2023 daemon.info avahi-daemon[3521]: Registering new address record for fe80::9944:a79c:3812:1e1b on tun163.*.
Mon Mar 20 14:22:24 2023 daemon.info avahi-daemon[3521]: Withdrawing address record for 172.19.163.1 on tun163.
Mon Mar 20 14:22:24 2023 daemon.info avahi-daemon[3521]: Leaving mDNS multicast group on interface tun163.IPv4 with address 172.19.163.1.
Mon Mar 20 14:22:24 2023 daemon.info avahi-daemon[3521]: Interface tun163.IPv4 no longer relevant for mDNS.
[... cut ...]
Mon Mar 20 14:22:24 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:24 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:24 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
Mon Mar 20 14:22:24 2023 daemon.warn dnsmasq[8758]: failed to create listening socket for fe80::4ad5:39ff:fe42:f70f%pppoe-WAN: Address not available
```
iStoreOS版本是: iStoreOS 21.02.3, 2023032001

jjm2473, 2023-03-20:
@NewbiePanda 我说的默认集成是kmod-tun，除非手动卸载了，不如不可能需要自己安装，而且有些设备还没法在线安装内核模块，例如rk3568方案的机型，一旦卸载就没了，只能通过重置来恢复。

jjm2473, 2023-03-20:
@Tanamaz 日志里面看不出uu为什么错误的，因为uu根本就没打印日志。需要找uu的技术人员来定位。

bcseputetto, 2023-03-24:
> @Tanamaz 日志里面看不出uu为什么错误的，因为uu根本就没打印日志。需要找uu的技术人员来定位。

/tmp/uu/uu.conf 里可以修改uu的日志等级

jjm2473, 2023-03-26:
@bcseputetto 但是不知道日志会打到哪里去

jjm2473, 2023-03-26:
@Tanamaz 现在istoreos还在21.02，不是22.03，可能不是同一件事

jazzylau, 2023-04-05:
我也是一模一样的问题，一加速就失败

jjm2473, 2023-04-14:
试试这个固件有没有问题 https://we.tl/t-F6F84Jyz6O

jjm2473, 2023-04-15:
最新固件应该修复了

closed 2023-04-15 by jjm2473

cero-ss, 2023-04-15:
> 最新固件应该修复了

测试最新固件，确实能够正常加速了，感谢修复，所以到底是什么原因导致的呢，好奇

jjm2473, 2023-04-16:
@cero-ss 之前的内核没开ipv4的策略路由，只开了ipv6的策略路由。更新日志有更详细的说明 https://github.com/istoreos/istoreos/discussions/713

lwj1994, 2023-06-13:
我主路由刷的 iStore ，然后路由器连接主路由，有 2 个问题
xbox 连接路由器总是显示 NAT 限制，游戏进不去。
uu加速器启动后，app 里找不到，还有什么额外设置吗？

cero-ss, 2023-06-13:
> 我主路由刷的 iStore ，然后路由器连接主路由，有 2 个问题 xbox 连接路由器总是显示 NAT 限制，游戏进不去。 uu加速器启动后，app 里找不到，还有什么额外设置吗？

机器需要连装有uu加速器的机器，哪个路由器装了插件，在xbox上就填上那个路由器的网关，然后建议把ipv6给关了

jjm2473, 2023-06-15:
@lwj1994 既然你有二级路由，那二级路由也要支持相应的NAT类型啊，或者upnp应该也可以。