istoreos/istoreos #588: R2S最新固件某种情况下无法启动

666dechaoge, 2023-02-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：更新r2s支持r2c的那次固件后，wan口接上级路由，lan口接下级无线路由器就无法启动，表现为红灯亮，两个网络灯都黑，无法进入管理页面

### (2) 路由器型号和固件版本

A：R2S  istoreos-21.02.3-2023020315-r2s-squashfs.img.gz

### (3) 详细日志和/或截图

A：无法进入管理页面

666dechaoge, 2023-02-14:
刷回20221216就可以了

jjm2473, 2023-02-14:
@666dechaoge 麻烦用R2S测试下这个固件 https://we.tl/t-mQKL9IoVKj ，并且启动以后执行 `dmesg | grep 'Set MAC tx_delay'` 看看输出

666dechaoge, 2023-02-14:
> @666dechaoge 麻烦用R2S测试下这个固件 https://we.tl/t-mQKL9IoVKj ，并且启动以后执行 `dmesg | grep 'Set MAC tx_delay'` 看看输出

٩(•̤̀ᵕ•̤́๑)ᵒᵏᵎᵎᵎᵎ，晚点我试试

666dechaoge, 2023-02-14:
找了个空卡全新刷入了，在上述wan口接上级路由，lan口接下级无线路由器的情况下，直接启动或者web端重启都正常

666dechaoge, 2023-02-14:
刷入后第一次启动

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.188 (build@buildserver) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 2022120817)) #0 SMP PREEMPT Thu Feb 9 06:10:15 2023
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] cma: Reserved 8 MiB at 0x000000003f800000
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA32 zone: 4088 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 21 pages/cpu s47832 r8192 d29992 u86016
[    0.000000] pcpu-alloc: s47832 r8192 d29992 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1004572K/1046528K available (7806K kernel code, 572K rwdata, 2396K rodata, 1664K init, 713K bss, 33764K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x3a0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000007] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001603] Console: colour dummy device 80x25
[    0.002071] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.003004] pid_max: default: 32768 minimum: 301
[    0.003684] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004363] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007355] ASID allocator initialised with 32768 entries
[    0.008012] rcu: Hierarchical SRCU implementation.
[    0.009676] smp: Bringing up secondary CPUs ...
[    0.010887] Detected VIPT I-cache on CPU1
[    0.010969] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.011808] Detected VIPT I-cache on CPU2
[    0.011871] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.012670] Detected VIPT I-cache on CPU3
[    0.012733] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.012857] smp: Brought up 1 node, 4 CPUs
[    0.016169] SMP: Total of 4 processors activated.
[    0.016601] CPU features: detected: 32-bit EL0 Support
[    0.017070] CPU features: detected: CRC32 instructions
[    0.025660] CPU: All CPU(s) started at EL2
[    0.026122] alternatives: patching kernel code
[    0.040491] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.041396] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.042693] pinctrl core: initialized pinctrl subsystem
[    0.044456] NET: Registered protocol family 16
[    0.047272] DMA: preallocated 256 KiB pool for atomic allocations
[    0.049004] cpuidle: using governor menu
[    0.049632] hw-breakpoint: found 6 breakpoint and 4 watchpoint registers.
[    0.050505] Serial: AMBA PL011 UART driver
[... 995 lines cut ...]
[   22.751559] r8152 4-1:1.0 eth1: carrier on
[   22.801550] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:00] driver [RTL8211E Gigabit Ethernet]
[   22.811779] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[   22.812496] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[   22.813116] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[   23.739781] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   24.238509] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   26.912893] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   26.913752] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   28.205896] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[... cut ...]
[   47.952551] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   47.952560] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   47.952566] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
```

上述两种情况```dmesg | grep 'Set MAC tx_delay'```结果都为空

666dechaoge, 2023-02-14:
> @666dechaoge 试试看这个 https://we.tl/t-pqbXsKJZCg ，应该会有 'Set MAC tx_delay' 的输出了

这次首次启动或者重启 ，grep都有结果了

首次启动截取

```
[   33.290451] rk_gmac-dwmac ff540000.ethernet: Set MAC tx_delay: 0x24 rx_delay: 0x18 for PHY 0x001cc915
```
 完整首次启动记录
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.188 (build@buildserver) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 2022120817)) #0 SMP PREEMPT Thu Feb 9 06:10:15 2023
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] cma: Reserved 8 MiB at 0x000000003f800000
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA32 zone: 4088 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 21 pages/cpu s47832 r8192 d29992 u86016
[    0.000000] pcpu-alloc: s47832 r8192 d29992 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1004572K/1046528K available (7806K kernel code, 572K rwdata, 2396K rodata, 1664K init, 713K bss, 33764K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x3a0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000007] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001610] Console: colour dummy device 80x25
[    0.002070] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.003006] pid_max: default: 32768 minimum: 301
[    0.003698] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004377] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007393] ASID allocator initialised with 32768 entries
[    0.008050] rcu: Hierarchical SRCU implementation.
[    0.009724] smp: Bringing up secondary CPUs ...
[    0.010915] Detected VIPT I-cache on CPU1
[    0.010997] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.011837] Detected VIPT I-cache on CPU2
[    0.011897] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.012696] Detected VIPT I-cache on CPU3
[    0.012756] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.012885] smp: Brought up 1 node, 4 CPUs
[    0.016196] SMP: Total of 4 processors activated.
[    0.016628] CPU features: detected: 32-bit EL0 Support
[    0.017098] CPU features: detected: CRC32 instructions
[    0.025694] CPU: All CPU(s) started at EL2
[    0.026155] alternatives: patching kernel code
[... 1009 lines cut ...]
[   22.671108] r8152 4-1:1.0 eth1: carrier on
[   22.717083] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:00] driver [RTL8211E Gigabit Ethernet]
[   22.717962] rk_gmac-dwmac ff540000.ethernet: Set MAC tx_delay: 0x24 rx_delay: 0x18 for PHY 0x001cc915
[   22.731455] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[   22.732169] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[   22.732785] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[   23.643537] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   24.198884] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   26.812642] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   26.813531] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   28.092226] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   29.541232] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   29.541974] NFSD: Using legacy client tracking operations.
[   29.542458] NFSD: starting 10-second grace period (net f0000040)
[... cut ...]
[   48.188427] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   48.188440] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   48.188450] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   48.188457] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.

```

jjm2473, 2023-02-14:
@666dechaoge OK，delay值没问题了。 
你说的重启没网的情况，试试完全断电以后，插网线启动，看能不能复现

666dechaoge, 2023-02-15:
> @666dechaoge OK，delay值没问题了。 你说的重启没网的情况，试试完全断电以后，插网线启动，看能不能复现

全新刷入的，可以正常启动，需要我安装我之前用的那些插件再测试吗

666dechaoge, 2023-02-15:
我把常用的插件都装回来了，包括openclash wireguard nps frpc（另外卸载了docker），可以完全断电后正常启动

jjm2473, 2023-02-15:
@666dechaoge  我改的那个只是修改delay值的，delay值就算有问题应该也不至于出现网卡识别不了的，所以感觉还是没修复。你用几天看看还会不会出现启动以后没网络吧。

666dechaoge, 2023-02-15:
> @666dechaoge 我改的那个只是修改delay值的，delay值就算有问题应该也不至于出现网卡识别不了的，所以感觉还是没修复。你用几天看看还会不会出现启动以后没网络吧。

好，那我用这个固件挂几天

colindog, 2023-03-04:
我也遇到了，最近比较忙，直接先把R2S下了

jjm2473, 2023-03-13:
@colindog 你遇到的是一启动就不能访问后台，还是一段时间后才不能访问？

closed 2023-03-16 by jjm2473