istoreos/istoreos #592: 关于树莓派4B启用SMABA,挂在硬盘，进不去目录的问题

GD2021, 2023-02-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
[image]

A：

### (3) 详细日志和/或截图

A：
[image]
[image]
[image]

GD2021, 2023-02-15:
[image]

GD2021, 2023-02-15:
苹果手机访问也失败
[image]
[image]

jjm2473, 2023-02-17:
1. samba默认是没有用户名密码的，如果要设置账号密码，可以在“网络存储”-“统一文件共享”里面配置，而且注意默认不能使用root用户名
2. windows里面应该是使用`\\192.168.100.1`，而不是`//*`

closed 2023-02-17 by jjm2473