istoreos/istoreos #596: 静态路由bug，设置静态路由后，能ping通，但无法访问网页和ssh。

ggllioio, 2023-02-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

https://github.com/istoreos/istoreos/issues/312
遇到同样问题，未解决关闭。

## 2. 详细叙述

拓扑和设备IP如图，2台蒲公英x1旁路组网。
[image]

首先测试windows添加静态路由，istoreos不做配置。测试可正常ping通sft1200以及访问网页（如图）。
[image]

删除win静态路由，istoreos添加静态路由。
测试win可ping通IP，但无法打开网页，也无法访问22端口。
但此时在storeos内，可正常curl到sft1200网页（如图）。说明istoreos内部路由功能正常。
[image]

### (1) 具体问题

A：以上可知，x1设备以及PC功能正常，istoreos静态路由功能异常。

### (2) 路由器型号和固件版本

A：istoreos-21.02.3-2023020316-x86-64-squashfs-combined-efi.img

### (3) 详细日志和/或截图

A：如上

jjm2473, 2023-02-17:
istoreos配置好静态路由以后，执行 `route` 看看

ggllioio, 2023-02-17:
> istoreos配置好静态路由以后，执行 看看`route`

[image]

jjm2473, 2023-02-17:
@ggllioio 你的电脑默认网关是什么

ggllioio, 2023-02-17:
> @ggllioio 你的电脑默认网关是什么

pc默认网关192.168.11.1。
正常情况路由应将192.168.8.0段地址网关指向x1（192.168.11.25）。

jjm2473, 2023-02-17:
哦，可能是你的wifi路由器不支持ICMP重定向导致的，或者为了防止ARP欺骗故意不响应，小米路由器经常遇到。打开istoreos的防火墙设置，勾选lan的`IP 动态伪装`，然后看看是不是解决了

jjm2473, 2023-02-17:
还有一个方法，打开istoreos的网络接口设置，设备栏，配置`br-lan`，在“高级设备选项”里关掉“发送 ICMP 重定向”，应该也可以解决。

两个方法都试试，然后反馈下

jjm2473, 2023-02-17:
还有，看看AP的后台，有没有关于ARP欺骗的的设置

ggllioio, 2023-02-17:
> 还有一个方法，打开istoreos的网络接口设置，设备栏，配置`br-lan`，在“高级设备选项”里关掉“发送 ICMP 重定向”，应该也可以解决。
> 
> 两个方法都试试，然后反馈下

测试勾选lan的IP 动态伪装，问题解决。
没想到是ap的锅。目前使用京东云亚瑟ap模式。后台没有arp相关配置。
感谢老板百忙之中协助排查故障~
建议可以把相关信息写入文档，方便以后他人遇到相同问题。
再次感谢~

jjm2473, 2023-02-17:
@ggllioio 也试试关闭  `ICMP 重定向` 能否解决，这个的性能应该好一些。`IP 动态伪装`是防火墙的功能，需要修改包（其实就是NAT）

ggllioio, 2023-02-17:
> @ggllioio 也试试关闭 `ICMP 重定向` 能否解决，这个的性能应该好一些。`IP 动态伪装`是防火墙的功能，需要修改包（其实就是NAT）

br-lan关闭 `ICMP 重定向` 不行，还是能ping通不能访问网页。

jjm2473, 2023-02-17:
OK，谢谢反馈

closed 2023-02-17 by jjm2473

EchoGroot, 2023-02-26:
istoreos->ap（小米路由器）->群晖（80端口默认 302 到 5000）
在istoreos终端中，可以 curl 群辉80端口，显示需要重定向访问5000端口，但是curl 群辉5000端口 就没反应。可以ssh 群辉 22端口。尝试访问小米路由器下的其他设备，带端口的服务访问都没反应。

勾选lan的IP 动态伪装，问题解决。

hnzhaihui, 2025-01-29:
那么问题来了，能ping通，但无法访问网页和ssh，那就无法进入istroeos关掉“发送 ICMP 重定向”啊，我是用N1刷的旁路由，也突然遇到这个问题了！

hnzhaihui, 2025-01-30:
> 那么问题来了，能ping通，但无法访问网页和ssh，那就无法进入istroeos关掉“发送 ICMP 重定向”啊，我是用N1刷的旁路由，也突然遇到这个问题了！起因是：改了FTTR后，当天访问是正常的，但是发现外网访问N1的ddns设置失效了，是因为在光猫里的dmz要设置N1的IP，设置完当时也正常，但是过了1,2天就不能访问N1了。刚才把DMZ主机设置删除后重启也不行。
> 
> ## 刚刚试了，内网PC通过外网DDNS的域名+端口竟然能打开N1的web管理页面，让我来看看，怎么能解决
> iStoreOS 22.03.6 2024032917 “发送 ICMP 重定向”关闭，和 勾选lan的IP 动态伪装这个试了没有用。

找到问题了，在istroeos里除了映射管理网页的80端口，还试着映射了1个BT的22223端口，这个没成功，在光猫里设置成功了，就没管istroeos这个了，结果就是他导致的。