istoreos/istoreos #598: 关于iStoreOS作为旁路由或者静态路由到蒲公英，内网设备通过AP或者WIFI路由器访问联网失败问题

jjm2473, 2023-02-17:
案例1：iStoreOS 自身可以正常访问外网，并作为旁路由；内网设备通过 WIFI 连接，并且默认网关指向 iStoreOS，并且能够访问 iStoreOS， 排除 DNS 问题以后，内网设备无法连接外网，或者翻墙情况下可以访问被墙网站，却不能访问国内网站。

案例2：内网有蒲公英等组网设备，iStoreOS 配置静态路由将目标网段路由到蒲公英，iStoreOS 可以正常访问目标网络；内网设备通过 WIFI 连接，并且默认网关指向 iStoreOS ，并且能够访问 iStoreOS，却无法访问目标网络。

以上都可能是因为你的 AP 或者 WIFI 路由器不支持 **ICMP 重定向**，或者为了防止 **ARP 欺骗** 而故意丢包。

采用以下方案之一解决（优先使用前面的方案）：
1. 检查 AP 的后台有没有 **ICMP 重定向**  或者 **ARP 欺骗** 相关的设置，启用 **ICMP 重定向**  或者禁用  **ARP 欺骗** 相关设置。如果找不到设置或者故障依旧，采取下一个方案。
2. 打开 iStoreOS 的 “网络” - “接口” 页面，切换到 “设备” 标签栏，点击 `br-lan` 右侧的 “配置...” ，在对话框切换到 “高级设备选项” 标签栏，取消勾选 “发送 ICMP 重定向”，保存应用。如果故障依旧，采取下一个方案。
3. 打开 iStoreOS 的 “网络” - “防火墙” 页面，滚动到 “区域” 配置段，勾选 `lan` 区域右边的 “IP 动态伪装”，保存应用。

相关issue：https://github.com/istoreos/istoreos/issues/294 https://github.com/istoreos/istoreos/issues/302 https://github.com/istoreos/istoreos/issues/312 https://github.com/istoreos/istoreos/issues/596

EchoGroot, 2023-02-26:
istoreos->ap（小米路由器）->群晖（80端口默认 302 到 5000）
在istoreos终端中，可以 curl 群辉80端口，显示需要重定向访问5000端口，但是curl 群辉5000端口 就没反应。可以ssh 群辉 22端口。尝试访问小米路由器下的其他设备，带端口的服务访问都没反应。

勾选lan的IP 动态伪装，问题解决。

jjm2473, 2023-02-27:
@EchoGroot 谢谢反馈

marine19811006, 2024-01-13:
全部设置了，还是不行，国内网站很慢，有时打不开

jjm2473, 2024-01-13:
@marine19811006 那估计跟这个问题无关，这个问题的情景下是完全访问不了的。估计你的问题是翻墙插件导致的，要么插件自身的bug，要么配置问题。

marine19811006, 2024-01-14:
> @marine19811006 那估计跟这个问题无关，这个问题的情景下是完全访问不了的。此问题发生的条件是，wifi路由器在lan区域，并且从wifi来的流量需要路由回到lan区域（例如旁路由）。估计你的问题是翻墙插件导致的，要么插件自身的bug，要么配置问题。

我把翻墙插件都关了也一样，因为一开始以为是翻墙插件问题，所以都关了，还是老样子，后来卸载了也没用

jjm2473, 2024-01-15:
如果是openclash的话，有时候关掉也还是有bug的。你还可以检查下ipv6，关掉ipv6可能就好了。

persiliao, 2024-08-01:
小米路由器(AX3000T) -> iStoreOS(iStoreOS 22.03.6, 2024071911) -> phone

以上办法都试过，iStoreOS 联网正常，终端访问外网正常，Mac访问iStoreOS正常，iStoreOS可以看到有网络流量，但是phone连上主路由Wifi无法上网，防火墙也设置过高性能模式，lan的IP 动态伪装也勾选了，高级设备发送 ICMP 重定向也取消了，实在没找到解决办法了

jjm2473, 2024-08-02:
@persiliao 恢复出厂设置，只用向导配置旁路由，然后勾选防火墙 `lan` 区域右边的 “IP 动态伪装”。看看还有没有问题

persiliao, 2024-08-02:
> @persiliao 恢复出厂设置，只用向导配置旁路由，然后勾选防火墙 `lan` 区域右边的 “IP 动态伪装”。看看还有没有问题

还是一样，联不上网络

jjm2473, 2024-08-03:
那你还是换个wifi路由器或者改成无线桥接

persiliao, 2024-08-04:
> 那你还是换个wifi路由器或者改成无线桥接

只能这样了，改天再试

KevinZjYang, 2024-08-26:
我就是旁路由（istoreos）科学上网，truenas网关设置为旁路由，发现国外的网站都可以访问，国内的就有问题了，mp里面所有的站点都无法访问，在truenas里面ping也是不通，但是去旁路由ping就没有问题。
PS：同样是设置网关为旁路由，pc就一切正常~
解决办法：去防火墙打开lan口的IP动态伪装。

duanshuizhiliu, 2024-11-11:
你好，我用istoreos弄旁路由，装了SSR PLUS+,机场地址也都订阅了，状态检测也都正常。但是我用手机连接这个旁路由的wifi 内网可以正常访问，但是外网就打不开 是我哪里设置错了吗？

wmawma, 2025-04-23:
我的问题更复杂 windows11 连接一模一样的网段 留学没有任何问题，但是mac m4 看一会视频就被限速到只有几百k，也不是断流 就莫名其妙的限速了。 同一时段windows一点问题都没有。